w3c/automotive #222: Remove "0105-getVss-no-path-success.html" test case?

wonsuk73, 2017-07-07:
This test assertion is as below:
When the client makes a request with a correct 'getVSS' action without specifying 'path',
'VssSuccessResponse' is returned in which entire VSS tree is enclosed.

By the way the last f2f meeting at Birmingham, there was the consensus to make "path" attribute as a mandatory in vssRequest message. Therefore it might be removed from test cases.

@aShinjiroUrata Am I right?

aShinjiroUrata, 2017-07-10:
Hi @wonsuk73 san

>  there was the consensus to make "path" attribute as a mandatory in vssRequest message.

It looks this consensus is already refrected on VISS spec, right?

If so, I will fix(or delete) the test assertion and test case content.

wonsuk73, 2017-07-11:
@aShinjiroUrata As far as I understood there was the consensus to have "path" attribute as a mandatory even that is empty in Burmingham F2F. Because that is more easy to implement and less confusing when it is used. @acrofts84 Am I right? Do you have comment want to add? :-)

acrofts84, 2017-07-12:
Hi All, 
Yes, we agreed to make the path mandatory for getVSS. As Wonsuk said, this is simpler. It also encourages the client to only request the data it requires - the VSS tree will be a significant data structure, so we should encourage the client to look at individual branches, rather than the entire tree where possible.

aShinjiroUrata, 2017-07-12:
@wonsuk73 -san, @acrofts84 -san,

Thanks for explanation.
I understand that, 
- 'path' is mandatory 
- and/but empty path is acceptable.

I'll fix(or delete) the test assertion/test content.

wonsuk73, 2017-07-18:
@aShinjiroUrata Thanks!

closed 2017-07-18 by wonsuk73