w3c/automotive #224: Faulty description of "value" in example in chapter 8 Message Structure

erikbosch, 2017-08-14:
There seems to be a copy/paste error in section 8, JSON Schema Definitions:

https://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html

```
 "value": {
            "description": "The path to the desired vehicle signal(s), as defined by the Vehicle 
              Signal Specification (VSS).",
            "type": "string"
        },
```

I assume it should be "The data value returned by the server. This could either be a basic type, or a complex type comprised of nested name/value pairs in JSON format." as defined in the table above

drkevg, 2017-08-16:
Planning to complete along with #91

peterMelco, 2017-09-05:
Since this is done in 91, I will close.

closed 2017-09-05 by peterMelco