w3c/automotive #227: implementation report

peterMelco, 2017-09-05:
Under the branch implementation-report there is an suggested example of adding such a document to the specification(s).  Lets try to decide on the format of this if any.

peterMelco, 2017-12-01:
Should we just add links to the github repos to a page published as an appendix to the specs ?

peterMelco, 2018-03-13:
I will ask Melco to add the implementation report for the implementation the Melco demoed in Birmingham

INRIX-paul-boyes, 2018-03-13:
Where to post these.  How to link?

INRIX-paul-boyes, 2018-03-13:
Who can add test cases? Urata-san has started, but does not have time due to current work to complete.  We need more test cases.  Here is where they currently are:

https://github.com/aShinjiroUrata/web-platform-tests/tree/dev-urata-vsss-test

aShinjiroUrata, 2018-04-16:
In test table, some words "FAIL" are replaced with "PASS" by wrong editor operation.
Fixed them.
I'm sorry for the mistake.

wonsuk73, 2018-04-16:
Urata,
Thanks a lot for the report!
By the way could you share the link for the page I can see it with browser.
Current link you shared is to be showed the code for the report.

Thanks!

Kind regards,
Wonsuk.

aShinjiroUrata, 2018-04-16:
Hi Wonsuk-san
> By the way could you share the link for the page I can see it with browser.
> Current link you shared is to be showed the code for the report.
This implementation_report.html is in "implementation-report" branch and not in "gh-pages" branch.

I'm not sure if this is ok to merge in gh-pages. 
Most of the document is written by Peter-san, so I'd like to know his opinion.
(by the way, thanks for the document template, Peter-san.) 

Anyway I create pull request for review.

PeterWinzell, 2019-11-11:
Close , old issue no current relevance

peterMelco, 2019-11-11:
see above

closed 2019-11-11 by peterMelco