w3c/automotive #232: Callbacks vs promises

dontcallmedom, 2017-09-18:
VIAS uses callbacks for its asynchronous operations, when most of the JS world has moved to promises (which also enables usage of the new `async`/`await` ecmascript keywords).

Is there any reason to prefer callbacks to promises?

aShinjiroUrata, 2017-09-19:
Hello Dominique( @dontcallmedom ) -san,
Thanks for the comment.

> Is there any reason to prefer callbacks to promises?

Honestly, I have to say that there is no strong reason.

Histroically saying, I took over the editor's task from my successor around April 2017 and originally it was callback style.
The topic of changing to promise style was once raised in a discussion but at that time it was not strongly pushed to the 'promise' direction.

At the time, I did not push to 'promise' direction as well with minor reasons such as
- I was not confident that `promise` will be the best practice for long time since there are other asynchronous syntaxes such as  `generator`, `async/await`. ( and we sometimes hear word like 'promise hell'.)
- API users can easily write a wrapper with `Promise` or other syntaxes to cover callback style API if he/she prefers.
- At the time(May to June of 2017) I was hurry to fix the spec for turning to CR since it was already late for the scheduled timeline (=move to CR on April2017).

After that, I had been busy for other task and not updating VIAS for sometime.
During the time I personally got a few comments like 'shouldn't we use promise?' and I have been 
 thinking about whether we should change to Promise at this phase.

@dontcallmedom ,
do you think changing to promise style is something like nearly mandatory?

dontcallmedom, 2017-09-19:
Mandatory is probably too strong of a statement, but I can't think of any recent asynchronous Web API that would still be using callbacks. Promises aren't perfect, but they have much cleaner semantics, and they play very well with `async`/`await`.

Naturally, VIAS can be wrapped into promises - but since VIAS is itself a wrapper to the protocol, this starts to be lots of wrapping :)

(note that I am personally unconvinced that a standard wrapper is needed at all - see #233)

aShinjiroUrata, 2017-09-19:
Thanks for the comment. That sounds reasonable to me.

I'd like to hear if there is other people who have opinion on this issue?

wonsuk73, 2017-09-20:
I am on the position with promise rather than callback. Because that is promitive JS feature for supporting async call and which provides a clear benefit for developing and maintaining the program code.

peterMelco, 2017-09-20:
@wonsuk73  Not being  a web developer could you show by a simpe example what you mean ?

wonsuk73, 2017-09-20:
@peterMelco There are many article to compare the two different approaches (callback and promise). Below article is good simple note for differences between them. 
http://smilyanp.com/callbacks-vs-promises-differences-simple-notes/

aShinjiroUrata, 2017-09-25:
@wonsuk73 , @peterMelco 
Thanks for the comment.

> I am on the position with promise rather than callback. 

I suppose this may be good timing to change `get()` and `set()` to promise style.
Please let me know if other members has affirmative/negative opinion.

sou3ilow, 2017-09-26:
+1 for promise where both are ok.
I'm not sure that all callback can be changed to promise.

aShinjiroUrata, 2017-09-27:
@sou3ilow -san, thanks for the comment !

anawhj, 2017-10-10:
I'm sure that all callbacks can be replaced with promises or other ways. Apart from @aShinjiroUrata's comments on why we chose the callbacks above, there seems a discussion in the last f2f meeting in [#198](https://github.com/w3c/automotive/issues/198#issuecomment-301224261).

I would like to put some comments to consider reasonable designs of VIAS. I hope we can finalize the work while properly solving the concerns on VIAS, even though it would be a WG note or a reference for the relevant library as raised in #233. I believe that VIAS is a valuable work in terms of making the web apps easily access the VIS server.

- Note that W3C standard API should be designed following the recent trends with strict and elaborate. Otherwise, it will lose public confidence. There seems no callbacks for asynchronous Web API these days in consideration of readability, maintenance, and developer friendly API so that the most async functions have been handled with promises even in JS libraries. ([more info](https://medium.com/@ejpbruel/the-drawbacks-of-callbacks-or-why-promises-are-great-5dedf2c98c67), for why promise, not callback).
- There has been some issues such as stream data that couldn't be handled by promise according to [the guideline](https://www.w3.org/2001/tag/doc/promises-guide) for using promise. `subscribe()` method is one of the examples because the method should always be run for waiting to receive push messages even after the method is called once. we can investigate a way to effectively handle the case in [Push API](https://w3c.github.io/push-api/#pushmanager-interface). In the spec, `subscribe()` method is handled with promise through reasonable mechanisms within the algorithm description.
- I encourage VIAS to use pre-defined IDLs made by W3C. For example, VIAS defines VISError dictionary with several attributes, but we could see the pre-defined types in [error names tables](https://heycam.github.io/webidl/#idl-DOMException-error-names) and just refer the names within promise algorithms. I wish we could try to use the existent terminologies and IDL as much as we could. [Presentation API](https://w3c.github.io/presentation-api/#terminology) would be one of the great examples to make the best use of terminologies from other references. In VIAS, there could be possible to re-use some IDLs such as Web Messaging, WebSocket.

aShinjiroUrata, 2017-10-10:
Thansk for the comment, @anawhj -san.

- Although, using promise for rucurrubg event seems not common method, I agree to investigate PushAPI's subscribe() method.
I suppose revising one shot APIs (such as ge, set) to promise style would be not controvertial and I'd like to revise VIAS on this point when I can work. 

- Regarding VIAS's error handling, I agree to the idea that use pre-defined IDLs is preferable. 
On the other hand, VIAS have to be able to handle Error variations defined in VISS. So I would like to look at VISS error definitions as well, when considering to use pre-defined error name tables.

aShinjiroUrata, 2018-01-08:
Excuse me to become so late.
I have updated oneshot methods to promise style with below commit.

https://github.com/w3c/automotive/commit/3bf1b36c99f8157ea04b010843a47d830724215a

Then, close this issue.

closed 2018-01-08 by aShinjiroUrata