w3c/automotive #235: CSS selector for paths

dontcallmedom, 2017-09-18:
The spec calls upon CSS selectors to select a path in the VSS tree, but it's not clear how to apply the CSS selection algorithms to a YAML or JSON tree. In particular, I assume only a subset of CSS selectors would be applicable, but that would need to be defined

aShinjiroUrata, 2017-09-25:
> In particular, I assume only a subset of CSS selectors would be applicable, but that would need to be defined

Actually I rasied similar issue in #137 as 
 H) 6. VSS Object > pathsByCSS()

and the conclusion at that time was 

> H) That is a great question, and I think the best bet now is to, as you said, leave it as an implementation-dependent feature set. I will add a note to that effect.

Since CSS specification is very large, to keep specification simple, leave as implementation dependend
 is, I thought good decision. 
Or, it would be also achievable to define minimum set of supported selectors
 (in my preference, I'd like to support following as minimum
    E F:  Descendant combinator
    E > F: Child combinator )
 and leave the rest as implementation dependent.

Hello @drkevg -san,
I think CSS selector like interface was proposed by Kevin-san.
Do you have any preference on which part of CSS selectors you'd like to have in spec?   

> The spec calls upon CSS selectors to select a path in the VSS tree,
>  but it's not clear how to apply the CSS selection algorithms to a YAML or JSON tree. 

Actually the reason for not cleary stated was, supported CSS selectors was implementation dependent.
If minimum spec is defined, hot to apply those CSS selectors could be described, probably by examples.

aShinjiroUrata, 2018-01-08:
Looks there is no more comment, close this.
Please reopen if someone want to continue.

tguild, 2018-01-09:
Lack of recent comment is not criteria for closure. This warrants further discussion and consensus before closing.

tguild, 2018-08-21:
VIAS is not presently continuing on REC track and was published as a Note.

closed 2018-08-21 by tguild