w3c/automotive #236: Relation to Generic Sensor / WoT Scripting API

dontcallmedom, 2017-09-18:
A lot of the API surface exposed in VIAS addresses similar needs to the ones that the Generic Sensor API on the one hand, the WoT Scripting API on the other try to address; in particular, the WoT Scripting API is meant to provide easy access to hosted services representing sensors and actuators, which seems very similar to what VIAS wants to achieve.

Is there any expectation to align with one or the other approach instead of creating an auto-specific approach?

aShinjiroUrata, 2017-09-19:
> Is there any expectation to align with one or the other approach instead of creating an auto-specific approach?

As personal opinion, I think it would be good to align with WoT API or Generic Sensor API when one of those become nearly defacto standard style or going to be that in high possibility.
However, although I'm not watching the recent acitvity closely, I suppose both API are still changing and also I imagine some possibility that there might be a discussion of aligning WoT API and Generic Sensor API.

To try to align with other spec which is still changing requires certain efforts and time and my anxiety is it might not be able to meet our charter schedule.
To meet charter schedule, I'd prefer to keep VIAS's current spec for now. 
And when WoT API or Generic Sensor API become stable, I'd like to consider to align with one of those as next version (or level two) of VIAS.

dontcallmedom, 2017-09-19:
yeah, schedule-wise that sounds reasonable. I think it would still be useful to see if there are shared lessons to learn from the two other specs. In the meantime, feel free to close this issue.

aShinjiroUrata, 2017-09-20:
Thanks @dontcallmedom -san.
Then I close this for now.

closed 2017-09-20 by aShinjiroUrata