w3c/automotive #241: subscribe test cases may result in unexpected failure

aShinjiroUrata, 2017-11-20:
Test case 0180-subscribe-success.html
(https://github.com/aShinjiroUrata/web-platform-tests/blob/dev-urata-vsss-test/vehicle/viss/0180-subscribe-success.html)
 may result in unxpected failure even when subscribe itself succeed and the test criterion is satisfied.

This happend in test suite execution with ETRI's VISS implementation.
This does not happened with ACCESS's implementation.

aShinjiroUrata, 2017-11-20:
The test failure can happen by following scenario.

1)Testcase0180 send subscribe request to VISServer
2)Receives SubscribeSuccessResponse and some SubscriptionNotification
3)Then Testcase0180 send unsubscribe request to VISServer
4)After that, Testcase0180 expects to receive UnsubscribeSuccessResponse
  and if receives any other response from VISServer, judge it as 'Failure'.
5)However, even after unsubscribe request has been sent to VISServer, 
   some remaining SubscriptionNotification can reach to client and this is not errorneous behavior.
   But this is judged as Failure by 4) logic.

This issue can be fixed by ignoring SubscriptionNotifications reached to client after unsubscribe request has been sent.

aShinjiroUrata, 2017-11-20:
Bug fixed by following commit
https://github.com/aShinjiroUrata/web-platform-tests/commit/2ad5d41f9c9305e89a500f7c274fcb48ba3740e1

wonsuk73, 2017-11-23:
@aShinjiroUrata Thanks for the work.
Right before I have done the test with your revised version of test cases. Unfortunately I got fail result like previous one. I will share the screen shots with you via email.

aShinjiroUrata, 2017-11-27:
I received report from @wonsuk73 that, test 0180 passed after below change.

https://github.com/aShinjiroUrata/web-platform-tests/commit/76e8918ca1df3ec906e742bc7807e74fc22068c4

aShinjiroUrata, 2017-11-27:
Also reported test 0200 fails with similar reason to 0180 failure which is
- Test0200 send several 'Subscribe request' to VISServer.
- Then test0200 wait for 'SubscribeSuccessResponse' or 'SubscribeErrorResponse'.
- However at this point, test0200 receives 'SubscriptionNotification' message and it judges as test failure.
  (Test0200 can ignore 'SubscriptionNotification' since it does not relevant to test criterion.)

aShinjiroUrata, 2017-11-27:
For test 0200, following change has been done.

https://github.com/aShinjiroUrata/web-platform-tests/commit/760c51b020fa09b4db6934aafc3bf40d1cbda5c0

wonsuk73, 2017-11-27:
Thanks @aShinjiroUrata ! It's working well!

aShinjiroUrata, 2017-11-28:
Thanks for the feedback.
In this issue, 0180, 0200 problems are reported, fixed and verified.
I close this issue.

closed 2017-11-28 by aShinjiroUrata