w3c/automotive #242: Error Info(number, reason, message) to subscribe request with filter for branch

wonsuk73, 2017-11-25:
In current  [9. Server Side Filtering](https://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html#server-side-filtering) part of VIS spec describe as below.

> Filters can only be applied for data which is represented by a primitive type, such as leaf nodes in the VSS tree, not entire branches. For example, a filter cannot be set on Signal.Drivetrain.InternalCombustionEngine.*, however it can be set on Signal.Drivetrain.InternalCombustionEngine.RPM.

By the way I couldn't find appropriate error number, reason and message in the [9.2 Errors](https://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html#errors) part. If it is true it should be added.

wonsuk73, 2017-11-25:
@acrofts84 @drkevg @aShinjiroUrata Does it make sence to you guys?

acrofts84, 2017-11-30:
Hi Wonsuk. 

I think the logical response to this would be a 400 Bad Request, but I agree that the error message "The server is unable to fulfil the client request because the request is malformed." doesn't provide a satisfactory response. 

How about introducing 400 Bad Request "Filter requested on non-primitive type"?

wonsuk73, 2017-11-30:
@acrofts84 That's good to me! But to be clear, error code is "400(Bad Request)" and message is "Filter requested on non-primitive type".So Error Reason need to be added. Right?

acrofts84, 2017-11-30:
Good point. I think the reason has to be a bit more concise, so I think the best approach is:

Code: 400 (Bad Request) 
Reason: filter_invalid
Message: Filter requested on non-primitive type

wonsuk73, 2017-11-30:
@acrofts84 Thanks!

closed 2017-11-30 by wonsuk73