w3c/automotive #245: [0300: filter range]

wonsuk73, 2017-11-27:
This test case is to check range filter for VIS spec as test assertion.
>Test assertion:
>If the client specifies the "Range" of the signal, the client receives data when the value satifies the conditions.

But it makes below subscribe request but above of range filter has higher value than below. It's wrong case for range filter.
>{"action":"subscribe","path":"Signal.Drivetrain.Transmission.Speed","requestId":"reqid-0290","filters":{"range": {"above":100, "below":20}}}

@aShinjiroUrata could you review and revise the test case?

aShinjiroUrata, 2017-11-28:
Thanks for pointing.
Use of 'above' and 'below' was opposite.
Fixed by below change.

https://github.com/aShinjiroUrata/web-platform-tests/commit/5ffab6e3833f271ad959ec2039b7be695597dd6c

##Actually, 'filter' function is not yet supported in ACS's VISServer and so testcases are not tested enough.

wonsuk73, 2017-11-28:
@aShinjiroUrata Thanks! By the way don't we need to have additional test case for filter error case?

aShinjiroUrata, 2017-11-28:
> By the way don't we need to have additional test case for filter error case?

I think it would be better to have.
Basically, I wrote test code according to test assertion document in #124.

When you have a test case to add, it would be helpful if you add an assertion to this document :-)
Although, I think I'll be able to do that by myself when I have time.
(I guess, it would be better to chage excel to markdown or something )

wonsuk73, 2017-11-28:
@aShinjiroUrata If you agreed, could you add that to test assertion doc?

Absolutely agreed with below opinion!
>I guess, it would be better to chage excel to markdown or somethin

wonsuk73, 2017-11-30:
@aShinjiroUrata I think this test case require 6 times satisfied repsonse for the range filters of subscribe request message. So it might be can make timeout even a server is working correctly. I think my VIS server is working well regarding to range filter condition, but it got the timeout now.

So  could you give long time than now like 20 sec for this test case?  To overcome the long duration issue with it you can add the logic which is test case do make fail right after server response with the out of range value for filter condition even that is one time.

What I understand is current TC request subs message with range filter, after that it try to receive message until six times then it make a result. If TC doesn't receive all of them until timeout duration, it say timeout!

What do you think?

aShinjiroUrata, 2017-11-30:
Hi @wonsuk73 

Thanks for the testing.
Timeout is one thing I was worrying about in this test system for vehicle.

I defined fixed timeout value '5sec' which is defined in vehicel-test-config.js as TIME_OUT_TIME
 and this timeout is applied to all the testcases.
I decided this value because it works with my VISServer but I did not expect
 this value fit for every VISServer implementation.

It is possible to use longer timeout time for this test0300, such as
 by defining special timeout "TIME_OUT_TIME_LONG = 20000;" and use this for test0300.

By the way, in 0300, vehicle speed is subscriibed and I supposed, six times of vehicle speed response
 would not take long time and 5second timeout would be enough.
In you VISS, how many times of vehicle speed response returns in one second? (or in ten second?)
It seems less frequent that I imagined.

wonsuk73, 2017-11-30:
@aShinjiroUrata Default notification period of my server is 1000ms. So I changed this value with 300ms. Then it works! So I think it might be need to have proper value for notification period of the server be tested. Am I correct?

aShinjiroUrata, 2017-12-01:
> @aShinjiroUrata Default notification period of my server is 1000ms. 
> So I changed this value with 300ms. Then it works! 

I think, since there is no neccesity to wait for 6 response, I should rewrite to wait for response
 by not a fixed number but a period of time.
Then even if your server's frequecy is low, test0300 receive not six but, say, three response and
 do the test with those three.
I'll do this.

> So I think it might be need to have proper value for notification period of the server be tested. 
> Am I correct?

I think you are right.
Probably we should add some Test Guideline to explain required condition for data source
 such as "vehicle speed should be issues more than once per 1sec.
(Data source, I mean, is vehicle data source connected to VISServer, such as real vehicle, simulator, etc.

aShinjiroUrata, 2017-12-01:
> I'll do this.

Change is done by below commit/push.

https://github.com/aShinjiroUrata/web-platform-tests/commit/0ca2ecc54a719a79943fdf45279f47c39703b1d4

@wonsuk73 , I expect that you will not meet timeout even if your notification period is 1000ms.
Please let me know when you have tried this.

aShinjiroUrata, 2017-12-15:
I close this since the change works OK in my environment.
Please reopen if you find any problem.

closed 2017-12-15 by aShinjiroUrata