w3c/automotive #246: [0190: receive the value continuously]

wonsuk73, 2017-11-27:
Urata, 
I guess previously this test case doesn't have issue. So it might be changed recently with small issue. " if (msg.action !== "subscription")" in below code might be wrong code. Because there is no action string as "subscription" in the spec so it return all of response message including subs notification.

@aShinjiroUrata Could you check this?

>    } else if (phase == PH_SUBSCRIBED) {
>      console.log("PH_SUBSCRIBED");
>      t.step_func(function() {
>        // wait subscriptionNotification
>        if (msg.action !== "subscription") {
>          addLogMessage("PH_SUBSCRIBED: Ignore messages except for Subscription notification.");
>          addLogMessage("msg = " + JSON.stringify(msg));
>          return;
 >      }

aShinjiroUrata, 2017-11-28:
Hi @wonsuk73 

> I guess previously this test case doesn't have issue. 
> So it might be changed recently with small issue. 

Yes, actually I changed this yesterday.

> " if (msg.action !== "subscription")" in below code might be wrong code. 
> Because there is no action string as "subscription" in the spec
>  so it returns all of response message including subs notification from server.

I was thinkgin so, and my VISServer imple was written so.
But when I looked at below spec, I noticed 'action':'subscription'
 is added in subscriptionNotification and so I changed the test 0190.

https://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html#dfn-subscriptionnotification

In the published draft, 'action':'subscription' should not exist
 but I guess following latest editor's draft is better.

wonsuk73, 2017-11-28:
@aShinjiroUrata Thanks for quick answer! I didn't know this change even I am one of co-editor for this spec ^^; I will change my implementation!

wonsuk73, 2017-11-28:
@aShinjiroUrata One question I have is if I add subscription action to my server implementation, do you think it works well with other test cases related with subs notification? I am just curious :-)

wonsuk73, 2017-11-28:
@aShinjiroUrata I have added subscription action to my server. It works well. and It seems there is no side effect for other test cases! Thanks!

aShinjiroUrata, 2017-11-28:
> do you think it works well with other test cases related with subs notification? I am just curious :-)

Actually, this change about subscriptionNotification message format is not finished and I will change some more.
It may affect to 180, 190, 260, 290, 300, 310, 320.
I will commit/push after all the change related to this fix has done
 and it works at least with my VISServer imple.

wonsuk73, 2017-11-28:
@aShinjiroUrata Okay! If you have done, let me know! I will test them :-)

aShinjiroUrata, 2017-11-28:
Hi @wonsuk73 

The change I mentioned above is done by below change.
https://github.com/aShinjiroUrata/web-platform-tests/commit/85d967062e04e4e6963272751f10dc2a4b97c1e1

Including minor fix, changed test are below.
180, 190, 210, 240, 250, 260, 280, 290, 300, 310, 320.

Test result with my VISServer is same with before this change.

wonsuk73, 2017-11-28:
@aShinjiroUrata Got it! Thanks!

aShinjiroUrata, 2017-11-29:
Hi, @wonsuk73 
If the test passed in your side, please close this issue :-)

wonsuk73, 2017-11-29:
@aShinjiroUrata I have done the test for 180, 190, 210, 240, 250, 260, 280, 290. It's fine!
But my implementation doesn't have enough implemenation for the last three test cases(300, 301, 302). For those test case I need to have sometime to check!

aShinjiroUrata, 2017-11-29:
Hi @wonsuk73 
Thanks for the feedback!

As this issue is originally for test0190, I close this issue.
If you find some problem in 300, 310, 320, please raise as new issue.

closed 2017-11-29 by aShinjiroUrata