w3c/automotive #247: Test case for subscriptionNotificationError

wonsuk73, 2017-11-28:
VISS has [subscriptionNotificationError](https://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html#dfn-subscriptionNotificationError) message format in the spec below. Don't we need to have test case for this? @aShinjiroUrata what do you think?

>{
>    "$schema": "http://json-schema.org/draft-04/schema#",
>    "title": "Subscription Notification Error",
>    "description": "Error message sent by the server when there is an error with an existing subscription",
>    "type": "object",
>    "required": ["action", "subscriptionId", "error", "timestamp"],
>    "properties": {
>        "action": {
>            "enum": [ "subscription" ],
>            "description": "The identifier for the subscription notification",
>        },
>        "subscriptionId": {
>            "$ref": "#/definitions/subscriptionId"
>        },
>        "error": {
>            "$ref": "#/definitions/error"
>        },
>        "timestamp": {
>            "$ref": "#/definitions/timestamp"
>        }
>    }
>}

aShinjiroUrata, 2017-11-28:
Hi @wonsuk73 

Actually, test 0230 in test assertion document is test for 'subscriptioNotificationError'.
(Please look at #124 's attached excel document)

However, I could not come up with a request to send to VISServer which can cause 'subscriptionNotificationError'.
So, eventually I did not write test 0230.

If you have good idea, please let me know.

wonsuk73, 2017-11-28:
@aShinjiroUrata Got it. About how to make a test case for subscriptionNotificationError, I don't know either. I will also think about that :-)

aShinjiroUrata, 2017-11-29:
> I don't know either. I will also think about that :-)

Thanks!
However, I think this might be difficult to write as automated test.

- If 'subscribe' is bad, just end up with 'subscribeError' and can not go to next step.
- So I suppose, to cause 'subscriptionNotificationError', after 'subscribe' succeed, some trouble should happen. 
- However client can not cause such trouble because what client can do is just send 'subscribe' request and send 'unsubscribe' request.
- Also, another difficulty is, this test code should be something that can certainly cause 'subscriptionNotificationError' for every VISServer implementation.

tguild, 2019-11-19:
we are closing outdated issues, please reopen or create a new one if still relevant

closed 2019-11-19 by tguild