w3c/automotive #249: [0310: filter minimum change]

wonsuk73, 2017-12-01:
The goal of this test is as below.
>Test assertion:
>If the client specifies the "Minimum change" of the signal,
>the client receives data when the value has changed by a specified amount.

Currnet test case do only one minChange test with subs request message below. My server doesn't support minChange until now. But It's passed two times. So I guess that it might be need few more test with other values like {"minChange":15} . 
>{"action":"subscribe","path":"Signal.Drivetrain.Transmission.Speed","requestId":"reqid-0310","filters":{"minChange":5}}

@aShinjiroUrata What do you think?^^

aShinjiroUrata, 2017-12-01:
> My server doesn't support minChange until now. But It's passed two times.

Same in my place :-)

> So I guess that it might be need few more test with other values like {"minChange":15} .

This is good idea. I didn't come up when I writing this test.
I'll do this change.

tguild, 2019-11-19:
we are closing outdated issues, please reopen or create a new one if still relevant

closed 2019-11-19 by tguild