w3c/automotive #253: Disposition of Comments on Direction of VIAS

Hirabayashi-KDDI, 2018-01-09:
As you know, there has been long discussion on "What should be included in the next Charter as the target for standardization." during and after the F2F meeting in Burlingame, and I'm very much confused about the current situation.
I'd like to discuss the rechartering topic with intention of including VIAS one of the targets for standardization during our telephone conferences in January.
In order to clarify what have been discussed so far and get a consensus about what to do for VIAS within the whole WG, the following "Disposition of Comments" paper has been summarized. 
Any comments are appreciated.
[Disposition of Comments on VIAS1.0.docx](https://github.com/w3c/automotive/files/1613686/Disposition.of.Comments.on.VIAS1.0.docx)

tguild, 2018-01-09:
I have removed the doc file since it included excerpts of conversations made in member-automotive mailing list contrary to W3C's Member Confidentiality. A copy is being sent to the mailing list.

https://www.w3.org/2017/Process-20170301/#confidentiality-levels

F2F, teleconference minutes on this topic have been public and there will be a public explanation of outcome regarding VIAS.

Hirabayashi-KDDI, 2018-01-10:
Thank you for your pointing out. I'm sorry for my careless mistake.
I will resend the removed document to member-automotive mailing list 
for a mutual understanding.

tguild, 2018-01-30:
This topic has been discussed in member confidential email thread, additional teleconferences and on a questionnaire open to the Auto WG

We reviewed the responses from the WG and the group formally resolved to publish VIAS as a Note.

https://www.w3.org/2018/01/30-auto-minutes.html

closed 2018-01-30 by tguild