w3c/automotive #256: 2018 draft charter changes

tguild, 2018-01-30:
Collect feedback from Auto WG on the draft charter

PatrickCQ, 2018-02-01:
Rephrasing of general scope: https://github.com/PatrickCQ/w3c-automotive/commit/4cf2ef5ad5c3a2f86fb176dc59eefb0707ec43eb

PatrickCQ, 2018-02-01:
Emphasis on creating a pattern for an interface and not a protocol.
https://github.com/PatrickCQ/w3c-automotive/commit/a3b9d3ca5b80272430cb0957fa0fcdb06cfd3b54

peterMelco, 2018-02-13:
Some comments:

1 Scope

 **Replace … with actual content or remove.**  
 
* Vehicle Data
    * vehicle brand, model, year, fuel type, transmission type, steering wheel position, tire pressure, oil level, wiper position, lights, doors, windows and seat settings as well as navigation, trip computer data, climate control data, speed, RPMs, acceleration, gears, …
* media
    * media control, track lists, …
* navigation
    * route manipulation, points of interests, …

2.1 Normative specifications

Should we define **“the vehicle universe”** or is that implicit ?

tguild, 2018-08-21:
Charter long ago published and approved by W3M and AC

closed 2018-08-21 by tguild