w3c/automotive #263: Review on how to start writing RSI specs at the beginning

anawhj, 2018-03-28:
In our group, the volunteers were gathered for RSI spec editors as the follows.
* RSI editors: https://www.w3.org/auto/wg/wiki/RSI_editors

About a year and half ago, VW submitted the proposal for RSI standardization to W3C, but it has been probably updated with a little changes so far. I guess VW would show us the updated version of RSI document at the f2f meeting. Before that, we can see the specification in the following link. (guessing no critical change for up to date)
* VW submission for RSI standardization: https://www.w3.org/Submission/2016/01/
(You can find the documents in automotive-bg/rsi github as well.)

In addition, there were a couple of RSI issue discussion in W3C automotive-bg github. Now, it could be discussed in W3C automotive wg github as the w3c automotive wg charter was revised even though it hasn't been fixed yet.
* automotive-bg github issues: https://github.com/w3c/automotive-bg/issues/
* charter extended: https://lists.w3.org/Archives/Public/public-automotive/2018Mar/0012.html

One more, there is a github repository for testing RSI server modules and for discussing the relevant issues.
* RSI server implementation: https://github.com/wzr1337/rsi.server
* Issues on RSI server: https://github.com/wzr1337/rsi.server/issues

RSI standardization has been already written by VW with a high level of maturity, so hopefully, we could easily make progress on the standardization with a few issues. I would suggest that we could discuss on how to make progress on RSI standardization using this thread first.

wzr1337, 2018-03-28:
I would like to add that I also provide a demo application that uses the rsi server under:

* RSI Demo Server using media plugins and server: https://github.com/wzr1337/rsi.demo

I am also working on a HTML5 client Demo and I will release it on github soon. I will POST a link here ;)

tguild, 2019-11-19:
we are closing outdated issues, please reopen or create a new one if still relevant

closed 2019-11-19 by tguild