w3c/automotive #265: Some bugs in current test cases

wonsuk73, 2018-04-17:
For preparing F2F meeting esp. regarding VISS implementation, I had recevied current version of test cases  from https://github.com/aShinjiroUrata/web-platform-tests/tree/dev-urata-vsss-test/vehicle/viss

And I found some bugs below. it seem it was made because of applying the spec changes as replacing getVSS with getMetadata in action and replacing vss with metadata in json schema for getMetadata action.
-  [0100: get a Vehicle Data Model tree with a valid getMetadata request]
  . GETMETADATA_STANDARD_PATH is not defined
  . '.vss' should be replaced with '.metadata'
- isMetadataSuccessResponse and isMetadataErrorResponse functions of vehicle-test-helper.js
  .  '.vss' should be replaced with '.metadata'
- [0105: get a VSS tree by getMetadata without specifying path]
  .  '.vss' should be replaced with '.metadata'
- [0110: get a VSS tree with a valid getMetadata request with a wildcard]
  . GETMETADATA_WILDCARD_PATH variable is not defined
  .  '.vss' should be replaced with '.metadata'
- [0120: get a VSS tree with an invalid getMetadata request]
  .  GETMETADATA_INVALID_PATH variable is not defined
  .  '.vss' should be replaced with '.metadata'
- [0300: filter range]
  . TIME_NOTIF_WAIT variable is not defined

@aShinjiroUrata Could you check this? If you don't have a time for this. I could make a PR for this.

aShinjiroUrata, 2018-04-17:
Hi @wonsuk73 
As for PR, I appreciate if you can do it.
However, I'll check it by myself anyway.

aShinjiroUrata, 2018-04-17:
Hi @wonsuk73 

The things you pointed should be implemented in below commit on 2017/12/1. 
https://github.com/aShinjiroUrata/web-platform-tests/commit/2361d584137579ffa8fe11e096ee0652aa4f082d

In my place, the test runs without this kind of errors.
Could you try again with latest version of "dev-urata-vsss-test" branch
in 'https://github.com/aShinjiroUrata/web-platform-tests' ?

wonsuk73, 2018-04-17:
@aShinjiroUrata 

The version you mentioned has the same issues. So I think you code still use 'vss' attribute name rather than 'metadata'.because then current code doesn't make  some of errors related with that.
Could you check that?

Below is the definition of the metadatasuccessresponse for getMetadata action.
https://www.w3.org/TR/vehicle-information-service/#dfn-metadatasuccessresponse

wonsuk73, 2018-04-18:
@aShinjiroUrata 
I have made five PRs to replace '.vss' with '.metadata'. Could you review and merge them?
When I checked issues of constant variable in vehicle-test-config.js, it's ok!

If you merge this then I will check again with that. Thanks!

aShinjiroUrata, 2018-04-18:
Hi @wonsuk73 .

Thank you so much!
I have merged the PRs.

> When I checked issues of constant variable in vehicle-test-config.js, it's ok!
Thanks, understood.

aShinjiroUrata, 2018-04-18:
I tested my VISS after fix with your changed test cases.
It worked well. 
Thanks!

tguild, 2020-01-07:
Closing old and seemingly complete issue. Please raise any current concerns as new issues.

closed 2020-01-07 by tguild