w3c/automotive #267: Inconsistency between VISS "timestamp" and VIAS "timeStamp"

colindefais, 2018-06-04:
Hello,

In the latest Vehicle Information Service Specifications (02/13/2018 and 04/18/2018), time stamps in JSON messages corresponds to the "time**s**tamp" property (with lower case "s"):
https://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html#dfn-timestamp

But, in the latest Vehicle Information API Specifications (06/21/2017 and 04/18/18), time stamps in objects  corresponds to the "time**S**tamp" property (with upper case "S"):
https://w3c.github.io/automotive/vehicle_information_api/vehicle_information_api_specification.html#dom-visvalue-timestamp

This inconsistency can lead to mistakes while trying to implement the specifications.
Especially since JSON objects can automatically be converted to JavaScript objects, the VIA developer can easily forget to change the property name "time**s**tamp" to "time**S**tamp" (believe my experience :wink:).

aShinjiroUrata, 2018-06-06:
Hi @colindefais 
Thanks for the comment.

The reason of choosing "timeStamp"(captial 'S') instead of "timestamp" is comming from "DOMTimeStamp"(https://www.w3.org/TR/2003/WD-DOM-Level-3-Core-20030226/core.html#DOMTimeStamp).
In case of "DOMTimeStamp", I suppose, "timestamp" is thought as consist from two words "time" and "stamp". 

However, as it looks "timestamp"(lower case 's') is widely used as DOM(HighRes)TimeStamp attribute, I suppose, better to change to "timestamp" for  consistency.
https://www.w3.org/TR/webrtc/#dom-rtcrtpcontributingsource-timestamp
https://www.w3.org/TR/generic-sensor/#dom-sensor-timestamp

I made #268 for this.

aShinjiroUrata, 2018-06-06:
PR merged.

aShinjiroUrata, 2018-06-27:
Since fix has been done, I close this.

closed 2018-06-27 by aShinjiroUrata