w3c/automotive #269: Trailing commas in JSON schemas

rndwb, 2018-06-07:
A lot of the JSON schemas in the Vehicle Information Service Specification have trailing white spaces and are therefore no valid JSON.
The comma always shows up after the "description" of "action" property.

gunnarx, 2018-06-07:
> A lot of the JSON schemas in the Vehicle Information Service Specification have trailing 
**white spaces** and are therefore no valid JSON.

1. You mean only trailing **commas**, as in the title, right?  (I assume trailing white space does not make invalid JSON, but to keep the document clean I think trailing white space should all be removed anyway.)

2. Could you fix it and send a change request?

gunnarx, 2018-06-07:
I agree, the **last** component in an array or map should not have a comma, so to clarify you are looking at this:

```
"action": {
   "enum": [ "authorize" ],
   "description": "The identifier for the authorize request",  <-- this comma must be deleted
},
```

rndwb, 2018-06-07:
> You mean only trailing commas, as in the title, right?
Of course. Commas was in my muscle memory, I don't know why.

> Could you fix it and send a pull request?
I can do that, but I thought it might be generated somehow.

> I agree, the last component in an array or (list of json objects) should not have a comma, so to clarify you are looking at this:
Correctly. It is all over the different schemas but only at this position.

gunnarx, 2018-06-07:
> I can do that, but I thought it might be generated somehow.

Actually these documents are written directly in **HTML**.  (See for example the [git blame](https://github.com/w3c/automotive/blame/gh-pages/vehicle_data/vehicle_information_service.html) - the different lines come from many commits). 

When looking at what you wrote I saw it could use a cleanup on general formatting & indentation.  But before putting that effort in, I think it would be good to agree on the preferred whitespace style.  (So feel free to fix the errors first).

rndwb, 2018-06-08:
> When looking at what you wrote I saw it could use a cleanup on general formatting & indentation. But before putting that effort in, I think it would be good to agree on the preferred whitespace style. (So feel free to fix the errors first).

I fixed the errors, but indeed there are some formatting issues left. By the way this is also true for the example code in the document. So yes, please agree on a style.

PR #270 (2018-08-17): Make the JSON schemas valid JSON by removing wrong trailing commas
This fixes #269.
closed 2018-08-17 by wonsuk73