w3c/automotive #281: Handling wildcard for HTTP requests

tguild, 2018-12-05:
VISS Web Socket requests can include wildcards to return collections of matching signals. In the next generation specification  being able to address signals over HTTP and potentially other transports the question is how to accommodate wildcards.

This was brought up on 2018-12-04 Auto WG call but no conclusion reached, hence opening issue for further discussion.

https://www.w3.org/2018/12/04-auto-minutes.html#item01

One thought was to draw from SPARQL querystring model.

joakimsod, 2018-12-06:
Sparql 1.1 supports query over HTTP protocol. 
In the following request, [QUERY] is the URL-encoded equivalent of this query, containing a wildcard:
 
select * {  
  <http://example.org#alice> <http://example.org#name> ?n   
} 

GET /repositories/test?query=[QUERY] HTTP/1.1  
Accept: application/json

tguild, 2019-11-19:
Supporting queries and with positioning change degrading wildcard usage

https://www.w3.org/2019/11/19-auto-minutes

closed 2019-11-19 by tguild