w3c/automotive #282: Payload consistency across transports

tguild, 2018-12-05:
The successor to VISS intends to handle additional transports besides Web Sockets, specifically HTTP but potentially others such as Unix Sockets as one OEM reported experimentation demonstrated substantial performance gain.

There is a strong desire to be consistent across the various protocols where possible in both communicating requests and in handling responses received so an application could conceivably be made transport agnostic. Using Web Sockets allows for essentially a custom protocol as was the case in VISS.  Some of the VISS methods such as subscribe do not translate readily to HTTP methods, set is a close match to POST.

There seem to be fewer obstacles for consistency in service responses. One suggestion was to make the body of the request the same despite also needing to use an appropriate HTTP header of method being used. Ted responded offhand that be believes that is not possible in HTTP, GET only sends headers and no request body, POST expects name/value pairs and PUT body content is pretty much open but expected to be a resource write operation. It may be possible for HTTP servers tosafely ignore request bodies for eg GET instead of throwing an error but would need to be confirmed. This might be a good question for the TAG or @ylafon. 

This was brought up on 2018-12-04 Auto WG call but no conclusion reached, hence opening issue for further discussion.

https://www.w3.org/2018/12/04-auto-minutes.html#item02

ylafon, 2018-12-05:
Note that POST is not restricted to key/value pairs in the payload, it depends on the media type used, although it is the case for `application/x-www-form-urlencoded`.

ylafon, 2018-12-05:
I also suggest taking a look at HTTP/3

wzr1337, 2018-12-18:
The POST should have a JSON body that represents the desired state in the same way it would be when retrieving it via GET or WebSocket Push. 

Any difference would mean additional cognitive load for developers and thus slow down adoption.

PatrickCQ, 2019-11-14:
In my opinion the to be transported payload shall be as fitting to the transport protocol as possible. Therefore the payload is not consistent over all transport protocols. 
E.g. pushing JSON in a MQTT topic does not seem right as MQTT has structural elements itself which should be used to use the transport protocol as native as possible.

tguild, 2019-11-19:
https://www.w3.org/2019/11/19-auto-minutes.html

while we are designing for Web Sockets and HTTP, we agreed to support other protocols and agree JSON not always the most appropriate

closed 2019-11-19 by tguild