w3c/automotive #301: Timestamp for value to cature time of update

rai20, 2019-03-25:
Hi,
I have implemented a version of W3C-VIS-Server which uses the VSS as the data model. It works fine but I think it would be helpful to add "timestamp" in the value response ( GET ,SUBSCRIBE) so that we know when the value was last updated. I face this issue in my W3C-VIS Server implementation, that the clients do not know when the Signal value has been last updated.  Does it make sense to add something like "updatetimestamp" or "lastupdate" tag in GET, SUBSCRIBE response ? or Is my understanding completely wrong?

gunnarx, 2019-03-25:
Hello @rai20.  Just jumping in with a different comment first, before the working group evaluates your change proposal (which I think would first affect the version 2.0 currently in discussion).

I just want to say that is great to hear because W3C (and GENIVI naturally) are interested in more implementations appearing of the specifications that are created.  For W3C in particular, several implementations it is a sign of acceptance and a requirement (I think?) for a specification to be formally approved.   For me, I think there is also the opportunity for more code sharing and improving one or a few fundamental implementations together.   

Is your implementation licensed with an open-source license by any chance?

rai20, 2019-03-25:
Hi @gunnarx . Yes the implementation is licensed under EPL 2 and the project is Eclipse Kuksa. Here is the link https://github.com/eclipse/kuksa.invehicle . Implementation is under w3c-visserver-api

gunnarx, 2019-03-25:
Oh, OK.  I was already aware of the Kuksa implementation, and I have also studied the code a bit :-)
Maybe later if we have time, I can share some more detailed thoughts on it.

tguild, 2019-11-19:
We don't have a good answer for this one unfortunately, extremely problematic

https://www.w3.org/2019/11/19-auto-minutes.html

peterMelco, 2019-11-20:
Closing this issue. The time a sensor value "changed" needs to be part of the atcual sensor data frame and synchronized against the the vehicle system clock. This is out of scope for this spec.

closed 2019-11-20 by peterMelco