w3c/automotive #306: Data Model confusion

danielwilms, 2019-06-12:
Hi,

I promised to summarize the whiteboard discussion, we had at the F2F. One main issue was the misunderstanding around the term data model. I tried to draw a picture on my understanding, excluded the term data model and tried to do some clarification in the terminology, so that we can clearly name, what we're talking about. I think it's crucially important that we get an alignment on this. So @UlfBj, @gunnarx, @tguild, @PatrickCQ, @klotzbenjamin, @rtroncy please have a look and let me know what you think.

[image]

gunnarx, 2019-06-12:
**Main content**

I feel like we are missing something.  I thought as first that **1 - Content**, was the actual concrete database of signal definition, and I felt then that whole description was understandable.  But looking closer "1" is described as a "simple graph" and in the example as a "simple sketch", so it is not the formal _detailed_ database of specific signal names and attributes then?   Where is that covered in the above description?

I don't fully know what the simple tree/sketch is for, and I feel like this needs something like a **4 - Data Definition**, with an arrow  coming from the specification to "4" which is tagged  "instance of" or "implements", or maybe there is a better word.

I was actually about to say that 1, 2, 3 in the examples do not relate to 1, 2, 3 in the previous Domain Taxonomy and should therefore not use the same numbers, but it seems you meant them to be examples? 

At least I think Your "2" in Domain Taxonomy is a specification defining the syntax rules.   But your "2" in Examples looks to me like an actual data definition, not a specification of syntax rules.  The examplelooks like something that may adhere to the rules of a specification, but it *is not* the specification?  Or how do you see it?

**Language and nitpicks**

This is not so much about the concepts, which we must fix first, but more to improve and make it even more clear.

--------

> Domain knowledge expressed in a domain taxonomy to bridge human and machine understanding.

How about "Domain knowledge \***is**\* expressed... 

----
To keep things as short and precise, let's work on conciseness:

* "logical and understandable" can be shortened to either "logical" or "understandable"

----
> The content is the agreement of the taxonomy

I feel this sentence can be clarified, but maybe that follows from my feedback on the concept above.

----
> how to formalize leafs and branches

leafs/leaves
Hmm... https://www.englishforums.com/English/LeafsVsLeaves/nvwxm/post.htm
Some dictionaries say both are OK.  In the most recent PR from Patrick, it was changed to leaves.

----
> how to formalize leafs and branches

Can "describe" be used instead of "formalize"?

How about "how to _describe_ leafs and branches?"
or did you maybe mean
"how to _formally describe_ ..."

danielwilms, 2019-06-12:
@gunnarx, thanks so much for the comments. I fixed the language issues.

I think it was more of a misunderstanding through the "sketch"-example. I fixed it now from content to data definition.

What I would like to separate is simply:
- Data definition as a domain description, what in my understanding most would call a data model
- The specification, as a result of a rule set on how to describe the data definition
- Tools and serialization

Number 1) is a bit artificial, but I think it's good to have a term for the data itself apart from specification and serialization. Updated picture atttached.

[image]

LaurentCQ, 2019-06-13:
Hi Daniel,

There is one thing that I find misleading, in so far that the infographic seems to imply that each domain taxonomy's data definition's specification can be described using its own rule set. I will assume that it is not the case and that we want to have a common rule set for all domains.

As I write this, I realise that the difference you make between "specification", "rule set" and "syntax for the domain taxononmy" is not clear to me so I would appreciate if you could elaborate a bit more on these points.

Still, I would assume that the "Rule set" is the closest thing in your inforgraphic to what I think should be a core part of gen2.

Cheers,
L

tguild, 2019-06-13:
As I commented on @PatrickCQ pull request, great to see you both arriving at some of the same terminology.
 
https://github.com/w3c/automotive/pull/305

As for what goes into the Gen2 specification,  we should agree or if necessary refine Patrick's wording influenced by yours. Given the confusion we caused ourselves final revision of infographic may be worth embedding. 

I think the readme landing page for VSS should be separated into the three pieces you have Data definition (vspec), Specification (VSS written more as a specification than it is currently) and a third page on tooling and serialization.

As Gen2 and VISS reference VSS, it would be helpful for that landing page (current readme.md) to be the VSS specification. We should probably further clarify for Gen2 we are expecting JSON. VSS specification itself may remain high level and leave the details in YAML, although could be worthwhile to have tooling generate detailed markdown for inclusion in readme.

danielwilms, 2019-06-14:
answering @LaurentCQ:
> here is one thing that I find misleading, in so far that the infographic seems to imply that each domain taxonomy's data definition's specification can be described using its own rule set. I will assume that it is not the case and that we want to have a common rule set for all domains.

I think in an ideal world that would be the case. Thinking of existing taxonomies, which should be supported that might be an issue. In this case the part of the Gen2 multi domain taxonomy specification could be the serialization of the source taxonomy mapped to the Gen2 reqs. The rule set describes the syntax, which leads to the specification ;)

> Still, I would assume that the "Rule set" is the closest thing in your inforgraphic to what I think should be a core part of gen2.

I agree. That would IMHO in the specification -> PR we were discussing the most during the last weeks :)

danielwilms, 2019-06-14:
answering @tguild 
> Given the confusion we caused ourselves final revision of infographic may be worth embedding.

Agree.

> I think the readme landing page for VSS should be separated into the three pieces you have Data definition (vspec), Specification (VSS written more as a specification than it is currently) and a third page on tooling and serialization.

I agree, and I would take the same infographic (domain taxonomy view). The thing I still would be seeing different is the data definition vs. specification. I see the data definition as the data structure and the spec as the vspec files. but as @gunnarx had issues with that as well maybe it causes more confusion than it helps. I just saw that within the discussions the difference between syntax and content seemed to be important. But maybe we could fine-tune it that we take over your interpretation and use Syntax for the second box and interpret the data definition as the vspec files. Any opinion on that?

gunnarx, 2019-06-14:
Daniel, and all.  First I'd like to just say, let's not rush toward a conclusion.  I think more work is needed here, and sorry for picking this apart in detail, but that is only because I want to make sure we don't end up in the same place again some time later.  I think we need to use precise, but also simple language as much as possible, and I would encourage you to do that.  We risk getting tripped up on the fact that most of the participants in this are not native English speakers (and the fact that this is not only English but very computer-engineering specific...).  OK, that was just a general comment.  I will continue:

gunnarx, 2019-06-14:
>> Given the confusion we caused ourselves final revision of infographic may be worth embedding.

> Agree.

Agree, *if* we manage to make it quite simple to understand.

gunnarx, 2019-06-14:
Some feedback is in reverse order, but I will quote to make each point.

Let me give an example of where we are not yet succeeding in communication, in my opinion:

>  I see the data definition as the data structure ...

It's not enough... we all bring different understandings of these words to the interpretation of what you say.

What did you mean by data structure here.  Does it include data-type (in computer science terms), and does it includes the other proposed meta-data of the VSS such as signal quality, sampling frequency, unit, ...?   Is it the entire collection of details of a leaf node in the YAML tree?

I would propose, instead of saying A is B, let's agree on which word is the closest to describe what you mean and then use that.  And then we also need to exemplify very clearly.  And give some description around it, what parts it includes and not.

https://www.merriam-webster.com/dictionary/structure
https://www.merriam-webster.com/dictionary/definition

> but as @gunnarx had issues with that as well maybe it causes more confusion than it helps. I just saw that within the discussions the difference between syntax and content seemed to be important.

Here you seem to imply (say) that I do not think it is important to make a difference between syntax and content, when I in fact think, and constantly say, that it **is very important**.  So, I must have been unclear and we are still not yet understanding each other.

gunnarx, 2019-06-14:
> existing taxonomies, which should be supported

Which ones are those?   

Taxonomy itself is a very generic word, but it's the best we can have here.  As a point of clarification however, are you envisioning that these different taxonomies have different syntax rules and different rules for which metadata needs to be included, and so on?     Or are they just different (sub-trees) that someone can come up with in the future, but still following the rule sets for content that we have put down in Gen2.

It would be good to point to the examples we have that cannot be brought into a single description of *how* to describe data.  Whatever the state of this is, In order to write a practically useful specification we have to work with examples of all those things we intend to cover, in my opinion.

>  In this case the part of the Gen2 multi domain taxonomy specification could be the serialization of the source taxonomy mapped to the Gen2 reqs. The rule set describes the syntax, which leads to the specification ;)

I have read this several times but do not understand it.  You did not perhaps mean _realization_ ?  I don't understand "serialization" in this context.   But I also don't understand the rest even with that change.   Then: "_the part of the Gen2 multi domain taxonomy..._"   Which part?  Please define the part... :-).   Unless you meant the whole first line of the infographic that is actually named "multi Domain Taxonomy," but then your statement does not make sense to me.    Did you mean **one part**?  Did you mean a specific taxonomy such as the one labled "VSS 2" on the second line of the infographic?

gunnarx, 2019-06-14:
Daniel, to make things even clearer.... What does this sentence mean?

> "The _content_ is the _agreement_ of the Taxonomy"

This is under Data Definition, so we are talking about the content of the Data Definition.

If I try to be more precise, is it OK to say the _content_ **is** the Taxonomy.  Then I think you are saying that it is the **agreed** content of the Taxonomy...
Sorry, these are language nitpicks but honestly it is too hard to understand if we don't fix it.
Because how can a Taxonomy agree on anything?  You mean the agreement of the committee/group of people that decided on the taxonomy?

Or are you saying that the _content_ is some agreed upon "derivative" (I mean that in a very general meaning) that comes from the Taxonomy?  

**Is** the (Data Definition) content the taxonomy, or a result of it?  Sorry, but that not 100% clear to me when you say it is the "agreement".   OK, I think you understand my point.  How can we write this better?

Or maybe we can skip the sentence completely if it doesn't clarify anything.  Also, I would say again that we can drop "logical and understandable" and focus on the words that really define it.  I think all of that could be just assumed, that we're aiming for something usable, high-quality, easy to understand, and so on.

UlfBj, 2019-06-19:
I like this, but I have (as I think several above) some problem with the terminology used, and in particular with how the word "specification" is used. In my understanding a specification describes the rules for how something can be built, which in this case would be the rules for creating the YAML files. If I have understood it right, In Daniels proposal above it is the YAML files that are named Specification, so what should they be called if it is used as I propose? "Domain specification" maybe?
So applying Specification on Data definition/Content results in a Domain specification, which can then trough Tools be transformed into a Domain serialization.

gunnarx, 2019-06-19:
Hello @UlfBj - good to hear from you.  I could not reach you on company email for another matter... could you send me an email or let me know how to contact you?    (My apologies for using this ticket for a private matter.)

gunnarx, 2019-06-19:
> If I have understood it right, In Daniels proposal above it is the YAML files that are named Specification

I believe the YAML files (in VSS) have _implicitly_ served both purposes.  In other words both defining data items and (implicitly) showing _how_ to define data items.

Before this discussion came up in W3C, it didn't seem to be a big problem to understand this.

For the purpose of this infographic and for formally defining terms, I agree however.  The concepts of  "how" and "what" need to be separated (as Daniel and others have also said).  But if examples are used to help the explanation then there must be separate examples of each one.  It cannot be done by example, if the example being pointed to serves more than one purpose.

danielwilms, 2019-06-23:
> If I try to be more precise, is it OK to say the content is the Taxonomy. Then I think you are saying that `it is the agreed content of the Taxonomy`

@gunnarx: the marked part of the sentence is precisely, what I wanted to say. I wanted to emphasise, that it's a released version of the content of the taxonomy. Maybe that would be an even better wording.

What would you say about it's the `It's the released content of the domain taxonomy`. 

> Sorry, these are language nitpicks but honestly it is too hard to understand if we don't fix it.

I fully agree and I'm grateful for the input!

>  I would say again that we can drop "logical and understandable" and focus on the words that really define it

I agree. My bad, It was a mistake when exporting.

danielwilms, 2019-06-23:
> Taxonomy itself is a very generic word, but it's the best we can have here.

To narrow that down I tried to define the content of what I see is a domain taxonomy in the graphic. Maybe that connection is not quite clear yet.

> are you envisioning that these different taxonomies have different syntax rules and different rules for which metadata needs to be included, and so on? Or are they just different (sub-trees) that someone can come up with in the future, but still following a single rule set for how to describe content?

I envision, that there is a rule set and syntax given by gen2, which is IMHO ideally as close as or identical to VSS2. VSS2 is set as main and domain taxonomy defining vehicle signals. The root node of VSS2 is the entry point of Gen2. 

Now we come to different domain taxonomies, which define domains outside of vehicle signals. In the F2F we had media or notifications as examples. The specification (in the sense of the graphic) of these domain taxonomies can be different, however their tools have to ensure, that their outcome applies to the rules set in gen2. In gen2 we have to think and define then, how it relates to the vehicle signals, as integration, ideally a subtree of some sort.

> It would be good to point to the examples we have that cannot be brought into a single description of how to describe data. Whatever the state of this is, In order to write a practically useful specification we have to work with examples of all those things we intend to cover, in my opinion.

I completely agree. But there I would like to ask support of @LaurentCQ or @PatrickCQ. The combination of different taxonomies was always a part I didn't understand in the PR we had up to now. My proposal would be:
1. Get the terms clear through this graphic
2. Define the rule set based on VSS2
3. Create an example with the media domain and iterate.

danielwilms, 2019-06-25:
Update of the info graphic
[image]

PeterWinzell, 2019-06-25:
Just a brief comment
The second paragraph we have information like datatype, etc... what is etc ? I want a specification to be clear. Can we write out what etc actually is or remove it ?

danielwilms, 2019-06-26:
@gunnarx @LaurentCQ @UlfBj another try after the discussion yesterday. I thought a bit about the right term for a specification replacement. I thought "Structure" could work, but on the other hand it would be confusing, as it could relate to data structure. So, I decided for Rule Set. Please vote up or come with suggestions, so that we can close the issue.

[image]

UlfBj, 2019-07-02:
Domain Definition instead of Data Definition?
In the text related to 3, in both Domain Taxonomy and Example I guess "specification" should be replaced by one of the above?

closed 2020-11-24 by tguild