w3c/automotive #310: Gen2 feature: Data model

UlfBj, 2019-10-23:
This chapter in the Gen2 CORE document needs to be updated to reflect that the data model to be used is VSS 2.0.

UlfBj, 2019-10-23:
I assign myself to the task of updating this chapter.

PatrickCQ, 2019-10-30:
Is it agreed by the automotive group that only VSS 2.0 is a valid datamodel for Gen2?

UlfBj, 2019-10-30:
To my understanding the WG has decided that VSS 2.0 is the data model to use in Gen2.

tguild, 2019-12-03:
this is partially resolved in https://github.com/w3c/automotive/issues/311 and https://github.com/w3c/automotive/commit/7f3ddfb2f8cf09b09b403f65ac6e5787dc03c10a but still some single tree leaning of https://github.com/w3c/automotive/issues/315 although softened with should instead of shall.

@UlfBj Ulf suggested on WG call being able to support both and leave to implementation to decide if it wants to add as a branch or link to external service. 

https://www.w3.org/2019/12/03-auto-minutes

I think we should focus on service registry as likely to influence this and suggest that as topic for next call and here in this issue.