w3c/automotive #316: Datatypes

danielwilms, 2019-11-09:
@UlfBj in #315 [here](https://github.com/w3c/automotive/pull/315#pullrequestreview-313423093)

> The datatypes Float and Double should maybe have references to their exact definitions, as these are not uniquely defined by their names only?

suggestion @wzr1337:
> In order to make it exact how about float32 float64 and float128 to express
float, double and decimal respectively?

wzr1337, 2019-11-09:
Are we talking about JSON transportation always? Then we do not even need to distinguish between the different float types as they will be sent as characters anyway. 
For a server and client implementation it might be interesting to now the bit size of the float, but we can also just omit overprecise digits or fill underprivileged ones with 0. 

JSON does not even know the difference between integer and number - float. Therefore the RSI spec (https://www.w3.org/Submission/viwi-protocol/) had to introduce this differentiation, as embedded software  clientsand servers - which this protocol was designed for in thefirst place - may need to optimize memory layout according to the expected values. But even in RSI/Viwi the number type is used for any type of floating point precision

UlfBj, 2019-11-09:
>> Are we talking about JSON transportation always?

Yes we are. So you are right, the float/double machine implementation is transparent at payload level. 
We can close the discussion on that. 
Thanks.

closed 2019-11-20 by UlfBj