w3c/automotive #319: Gen2 feature: Filtering

UlfBj, 2019-11-21:
This chapter in the Gen2 CORE document needs to be updated.

UlfBj, 2019-11-21:
I will push a PR with an update proposal.

gunnarx, 2019-12-05:
For reference, the mentioned PR is #320. 
 
(I want to propose that we try to handle either tickets or PRs but not both for the same issue (I mean if the ticket does not add anything, like here).  

In GitHub they are somewhat equal -- they use the same number scheme.

Having everything visible in the same place helps but rather than having a list I would then propose to enable the project management and use the Kanban board it provides.   It handles tickets and PRs equally, so you don't need to have duplicates of the same thing.  This matches also the previous request to work with a typical agile process, where it can be highlighted which tickets/PRs we prioritize at this moment @peterMelco)

UlfBj, 2019-12-05:
I agree that having things that are related on more than one place is not ideal.

However, the working model that we decided on in WG, to my understanding, was the following:
1. Create an issue on something to be worked on in the spec. 
2. Someone adds a comment to take on the task.
3. The result of the task results in a PR.
4. A discussion might start on the PR thread to sort out major objections.
5. The PR is merged and closed.
6. Remaining minor objections continue to be discussed on the Issue thread, and resolved as new PRs.

So when the PR is open the discussion should be held there, before and after the discussion is held on the Issue thread. 
I think it works, but maybe it can be improved.

UlfBj, 2020-06-01:
This was implemented in PR#320.

closed 2020-06-01 by UlfBj