w3c/automotive #321: Issue of test framework and cases for VIS spec

wonsuk73, 2019-12-12:
There is the [integrated test framework with VISS test cases](https://github.com/aShinjiroUrata/web-platform-tests/blob/dev-urata-vsss-test/vehicle/README.md), which is mostly contributed by Urata-San from Access. Recently there is some installation issue for this. So I think we need to use [recent version of web test framework](https://github.com/web-platform-tests/wpt) of W3C. I guess easy way is to put our test cases into current web test framework in github with creating new folder. Then if anyone would like to use VISS test cases, it only need to install web test framework of W3C via installation guide. What do you think?

For doing this I have two questions below.
1. Is there any process or guide to put test cases into W3C web test framework?
2. Are current test cases enough for VISS spec?

tguild, 2020-01-14:
Hi @wonsuk73 

We should probably create a folder here as you describe and fork Urata-san's test repo here as it will be easier for tracking issues and encourage wider maintenance. If you're inclined please do so.

Regarding your first question,  I'm not sure if the W3C web test framework will work for us as it is designed for a browser run time.

As to the second, I think the tests are largely sufficient and could potentially grow over time based on feedback.

tguild, 2022-03-08:
@wonsuk73 we are going to close this issue since VISS v1 is not going to advance in the Recommendation process as we are more focused on v2.

https://lists.w3.org/Archives/Public/public-automotive/2022Feb/0006.html

Are you interested in starting a test suite for us on v2, perhaps forking Urata-san's as a starting off point?

closed 2022-03-08 by tguild