w3c/automotive #325: Gen2 versioning vs VSS versioning

UlfBj, 2020-03-25:
I'd like to clarify what "exactly one VSS version" means here.

Is it OK to have minor or patch release upgrades, for example?  Those would presumably be compatible in format, but changed their content (signal list) slightly?

And I imagine the defined protocol version, which is very important to agree on in this exchange, might also be used with some proprietary extensions of the VSS signal list?

So, is the intention to lock down the _format_ of the VSS, or also content?  Example of incompatible format changes could be extensions that have been made in version 2,  such as instances [1..2], arrays, etc.  These VSS _capabilities_ must be negotiated by the protocol first of all.  But is there also more?

_Originally posted by @gunnarx in https://github.com/w3c/automotive/pull/323_

tguild, 2020-09-15:
@UlfBj to try to reflect thinking from today's call on catalog expectation

https://www.w3.org/2020/09/15-auto-minutes.html

UlfBj, 2020-09-21:
In the TRANSPORT spec, Websocket Intitialization ,chapter 5.2.1.1, it says:

"The sub-protocol name SHALL be 'gen2' with a version number suffix, e.g. gen21.0. The sub-protocol version will be associated with exactly one Gen2 Server Specification version so that the client and server can correctly validate and parse request and response message packets.
var vehicle  = new WebSocket("wss://wwwgen2", "gen21.0");"

This allows the client to request a protocol version for WS, but it maybe it should say something about the case where the server cannot serve the request. 
For HTTP there is a similar initialization chapter, but it is currently empty. 

Regarding VSS version handling, there is not anything explicit about it in the spec currently. 
The CORE spec, Data Model ,chapter 4, refers to the VSS documentation , http://genivi.github.io/vehicle_signal_specification/.

closed 2020-11-24 by tguild