w3c/automotive #326: Is there not redundancy in providing number, reason and message?

UlfBj, 2020-03-25:
Is there not redundancy in providing number, reason and message?
In this particular case, number and reason are equivalent (I think?), meaning that 404 **means** "not found", right?

_Originally posted by @gunnarx in https://github.com/w3c/automotive/pull/323_

gunnarx, 2020-03-31:
Another aspect is that there are several different reasons tied to the same HTTP status code.
There are 2 different types of "404" but  and for example 401 (Unauthorized):

```
401 (Unauthorized) | user_token_expired | User token has expired.
401 (Unauthorized) | user_token_invalid | User token is invalid.
401 (Unauthorized) | user_token_missing | User token is missing.
401 (Unauthorized) | device_token_expired | Device token has expired.
401 (Unauthorized) | device_token_invalid | Device token is invalid.
401 (Unauthorized) | device_token_missing | Device token is missing.
401 (Unauthorized) | too_many_attempts | The client has failed to authenticate too many times.
401 (Unauthorized) | read_only | The desired signal cannot be set since it is a read only signal.
```

For VISS/WebSocket the error is transferred in the JSON document:
```
{

    "action": "get",
    "requestId": "8756",
    "error": {"number": 404, "reason": "Not found", "message": "No match for path in tree."},
    "timestamp": 1489985044000

}
```
but the REST (plain HTTP) transport I think it should be clarified how this subtype can be communicated to the requester, and if it can.

My gut feeling is that we can't or shouldn't rely on that a "document" (JSON encoded info) is delivered when a request fails on HTTP.  Some web experts might weigh in on that however.

A way to resolve this is to study the HTTP specification to see if we can map this to any existing concept but the only thing I found was the "Reason-Phrase" in RFC 2616.  

**Chapter 6.1.1 "Status Code and Reason Phrase"**:

   The Status-Code element is a 3-digit integer result code of the
   attempt to understand and satisfy the request. These codes are fully
   defined in section 10. The Reason-Phrase is intended to give a short
   textual description of the Status-Code. **The Status-Code is intended
   for use by automata and the Reason-Phrase is intended for the human
   user**. The client is **not required to examine or display the Reason-
   Phrase.**
```
This sounds to me that there is a one-to-one mapping between code and phrase.  In other words 404 would always mean "Not found" and not give any additional information?

UlfBj, 2020-04-01:
>> It also says: The reason phrases listed here are only recommendations -- they MAY be replaced by local equivalents without affecting the protocol.

Formally I would say that the sentence above makes possible to define proprietary (non-RFC2616) reason phrases. 
The error message field seems harder to align with both transport protocols. So maybe that should be removed.

gunnarx, 2020-04-15:
> > It also says: The reason phrases listed here are only recommendations -- they MAY be replaced by local equivalents without affecting the protocol.

> Formally I would say that the sentence above makes possible to define proprietary (non-RFC2616) reason phrases.

Yes, clearly.  This is why I mentioned it.  But is the intention of this text to allow for simple translations / alternatives for some systems but still having consistently the same phrase for each error code?    My question is if it is your interpretation, and intention, that the system can deliver different phrases for the same error code.  In other words, the phrase is used to convey the error "subtype", i.e. the real reason for the error?

> The error message field seems harder to align with both transport protocols. So maybe that should be removed.

Does that mean no possibility to get the more detailed error information, or is it still possible?

UlfBj, 2020-04-16:
>>  My question is if it is your interpretation, and intention, that the system can deliver different phrases for the same error code. In other words, the phrase is used to convey the error "subtype", i.e. the real reason for the error?

Yes, that is my interpretation and intention. It is also the pattern used in VISS.

>> Does that mean no possibility to get the more detailed error information, or is it still possible?

The error phrase would then be the more detailed error information.

gunnarx, 2020-04-21:
> My gut feeling is that we can't or shouldn't rely on that a "document" (JSON encoded info) is delivered when a request fails on HTTP. Some web experts might weigh in on that however.

I am considering to reverse my opinion on the above statement.  After all, many (most?) web sites deliver a custom page when a 404 happens, and sometimes for others too.   I just do not know enough about HTTP to know if it reasonable to think that all failed attempts will be accompanied by a document, but  maybe that is OK.

tguild, 2020-05-26:
Ted to talk with @ylafon

tguild, 2020-09-15:
resolved to provide to provide message as payload in HTTP to be consistent with web sockets

https://www.w3.org/2020/09/15-auto-minutes.html

UlfBj, 2020-09-21:
PR#340 contains update to align it with the resolution mentioned in previous comment.

closed 2020-11-24 by tguild