w3c/automotive #327: Timestamp format

UlfBj, 2020-04-08:
Gen2 has currently the following timestamp definition, inherited from VISS:
timestamp | integer | The Coordinated Universal Time (UTC) time that the server returned the response (expressed as number of milliseconds).
This is not the same as Unix time " the number of seconds that have elapsed since the Unix epoch, that is the time 00:00:00 UTC on 1 January 1970, minus leap seconds."(Wikipedia). 
I think we should adopt a commonly accepted timestamp format, which I do not believe that the current is?
Should it be integer based (e.g. Unix time), or text based (e.g. ISO 8601 or RFC 3339)?
What time resolution is adequate, seconds or milliseconds?

gunnarx, 2020-04-15:
I see no other comments so I'll jump in with a few.

> Should it be integer based (e.g. Unix time), or text based (e.g. ISO 8601 or RFC 3339)?

Since the current encoding of responses is defined to be JSON and not particularly focused on being compact, I would say "text based" is acceptable.  As long as it is not too verbose.

A totally different encoding is required to be bandwidth efficient anyway.

It seems these are worthy goals in this environment:

* Following a standard (of course, and all of your proposals do)
* Easy to process
* Human-readable
* Not wasting **too many** bytes

Assuming you are speaking about an absolute "wall clock time" then the following format is ISO8601 compliant:

```
2020-01-10T02:59:43.492750Z
```
... which means there should be plenty of library support for reading and converting.  But even without library support, this is so regular that it is very easy to parse with custom code.   Libraries often handle more [difficult formats](https://stackoverflow.com/questions/10494312/parsing-time-string-in-python) like "Wed April 16 13:58:00 +0200 2020" but those are a lot more difficult to handle in custom code.

That is therefore my proposal for now.  Interesting to hear other people's opinion.  Also, if the format is limited to Zulu Time only (the Z at the end) then any parsing of time zones goes away too.

Interestingly there is a W3C document on this already:
https://www.w3.org/TR/NOTE-datetime .  It is a NOTE, not a specification.

> What time resolution is adequate, seconds or milliseconds?

I would say, perhaps microseconds

ISO 8601 allows several variations.  In this case I find it beneficial if it were limited to only the variation described above.

UlfBj, 2020-04-16:
I think you are spot on here @gunnarx. 
Your proposal have my support.

MagnusGun, 2020-04-22:
I agree with Gunnars comment since we anyway use json we should have it as text and in human readable format (ISO8601). 

Just a side note, do we really need microsecond precision? as the spec is currently written the timestamp comes from when the server sends its response and for me it doesn't make so much sense to have higher precision than milliseconds unless we also mandate some kind of timesync scheme between server and client.

Anyway, as mentioned above ISO8601 does not specify precision so maybe we don't need to either, but just rely on the client to be able to handle higher precisions if its available.

gunnarx, 2020-04-28:
I proposed microseconds mostly to be potentially compatible with other parts of the system, that might have this level of precision.  I always like to see standards that are applied as widely as possible to avoid having to write boring translation code.

On the other hand, I interpret the general ISO standard such that it  is possible to provide as many or as few decimals as you want in the time stamp.  So maybe it is enough to make a note of that.

tguild, 2020-05-26:
+1 to ISO 8601 used extensively at W3C for many years cf:

https://www.w3.org/TR/NOTE-datetime

We use it internally and an advantage is alphabetical sorting is also chronological

tguild, 2020-05-26:
https://www.w3.org/2020/05/26-auto-minutes.html#item06

UlfBj, 2020-06-04:
PR#329 merged, which closes this Issue.

closed 2020-06-04 by UlfBj