w3c/automotive #328: Update and rewrite access control chapter in CORE document.

UlfBj, 2020-05-07:
At the WG meeting May 5 it was decided that the Access control chapter (Authentication and Authorization) shall be updated, using the two sequence diagrams presented at the meeting as the base line. 
The existing text in the chapter comes from the VIWI access control solution, and is quite similar to one of the proposed flows, which should open up for reuse.

peterMelco, 2020-08-17:
Review of  Access Control Chapter.  This is a mix of remarks and questions.  General comment throughout the document  is to use normal capitalization .  Headlines should capitalize **A**xxxx **B**aaaa **C**ccccc.

6.1 Architecture

_1 Paragraph_
**The Gen2** ... -> Gen2 should be replaced later with a more suitable name

_2 Paragraph_
Client
**An application making protected resource requests on behalf of the User and with its authorization.**
Rephrase: An application making protected and authorized resource requests on behalf of its user. 

**Ecosystem manager**: Why does it have to be an ecosystem manager , is it not enough to just simply state system manager

6.2 Protocol flows

_1 Paragraph_

**AGS** should be written out and defined before we can use its acronym.

6.3.1
Remove: **of the vehicle the Client wants to have access to.**

6.3.3
Define **fresh** proof

**Purpose**  Short name of a purpose ? Is it an identifier or description. refer to purpose definition

6.4.1
Is **Delegated** or Delegating ?

6.4.2 
Remove ...**some kind of**  its unnecessary

6.4.3

rephrase: **the Client must also include in the request a fresh proof of possession** the client must also include a fresh proof of possession in the request.

6.4.4

**associated to** vs associated with... personal preference for with

6.5.2
Difficult to read the string 
**Public key|W3C...**
Suggestion:
Use the example:
**<public key> | "W3C_Gen2" | <exp>...**

6.6

Maybe rephrase or break up the sentence:
**From a security point of view...**
A client should follow the principle of least privilege, which means that a client should not have access to more signals than necessary 

6.6.1

How is the role **undefined** used  ?

6.7.1

**Purpose list**

The concept of a purpose list that pre development maps against use already defined use cases ? or on what premises is such a purpose list defined by the system manager ? I don't see how we could use such a concept for  3rd party development, or am I missing something here. Is there a process where the developer(s) is able to add purposes to the purpose list ? Does he want to ? 

6.7.2

Does the entire request fail if a subset if its signal request is found within the Scope list (no_access) ?

UlfBj, 2020-08-18:
>>Ecosystem manager: Why does it have to be an ecosystem manager , is it not enough to just simply state system manager

It comes from the established(?) naming "PKI ecosystem". 

>>Purpose Short name of a purpose ? Is it an identifier or description. refer to purpose definition

"Purpose" refers to items of the type found on the Purpose list. 

>>Difficult to read the string
Public key|W3C...

You are not reading the latest version, where this is removed.

>>How is the role undefined used ?

It is undefined it seems...
It could be used in the Purpose list and Scope list as a replacement of all roles (per sub actor). 
It could be used by an Admin Client to get maximum scope access. 
It could be deleted.

>>The concept of a purpose list that pre development maps against use already defined use cases ? or on what premises is such a purpose list defined by the system manager ?

The Purpose list is created and managed by the Ecosystem manager (which typically will be the OEM). It gives the Ecosystem manager control over which use cases that should be available, to which Roles, and the set of signals that should be accessible for this use case.

>> I don't see how we could use such a concept for 3rd party development, or am I missing something here. Is there a process where the developer(s) is able to add purposes to the purpose list ? Does he want to ?

 It is up to the Ecosystem manager to establish the rules for how the Purpose list can be updated. The details of this process I think is out of scope in the specification document, maybe it can be found in a Best practices document?

>>Does the entire request fail if a subset if its signal request is found within the Scope list (no_access) ?

My view is Yes. The Client should not "know" about the signals set as no access on the Scope list, they shall not be returned by the server on a service discovery request.

closed 2020-09-15 by tguild