w3c/automotive #331: GEN2 Review Chapter 4. Data model

peterMelco, 2020-06-08:
4. Data model

The **service** is intended for use with a tree-like logical taxonomy to represent the vehicle data. An illustrative example of such a tree structure is shown in Figure 1. While it is meant to support conforming taxonomies it was created principally with the Vehicle Signal Specification **(VSS)** in mind. For more details, see the VSS documentation.

**service**
We are introducing service here. Should that not be done further up or is that implied by the name VISS ?

**(VSS)**
remove or link 

Figure 1 Diagram showing an example VSS 2.0 tree.
So, we need an actual diagram here.

tguild, 2020-09-15:
merge with #306

peterMelco, 2021-01-21:
resolved.

closed 2021-01-21 by peterMelco