w3c/automotive #333: Gen2 name

tguild, 2020-06-16:
I want to start moving us toward naming Gen2. Please provide suggestions and if we do not have a clear candidate will setup a survey to decide.

An obvious one is VISS2 although that would in some ways seem wrong if additional data taxonomies besides VSS end up being exposed through it. VISS was solely for VSS whereas Gen2 nominally supports additional data taxonomies.

UlfBj, 2020-07-31:
Connected Vehicle Specification

adobekan, 2020-08-03:
Specification of connected vehicle interfaces -> SCENE, SCENT, PIONEER etc.  :)
Connected vehicle interface specification -> CVIS, CENTRIC, VISION, CHIP, CONVERSION etc.

I have tried https://acronymify.com, for generating acronyms in this case.

peterMelco, 2020-08-12:
I like Connected Vehicle Specification ... the acronym CVS becomes somewhat confusing though :)

wonsuk73, 2020-08-24:
Connected Vehicle Specification looks quite coool as a name of the spec! But its meaning looks very broad. So I think it might be better if we add an more focused word for the spec to the name. How about Connected Vehicle Interface Specification, CVIS :)

Or If we are willing to develop additional specs related with Connected Vehicle, I might need to think about Series spec. Then Series style name could be better like below.
—> Connected Vehicle - Core Specification
—> Connected Vehicle - Transport Protocol Specification

UlfBj, 2020-09-22:
With the service catalog initiative in mind, maybe the following could show the relationship between the two.

Connected Vehicle Signal Specification (CVSiS) - the signal interface (current Gen2) 
Connected Vehicle Service Specification (CVSeS) - the service interface

Vehicle Signal Specification (VSiS) - the signal data model and catalog (current VSS)
Vehicle Service Specification (VSeS) - the service data model and catalog

tguild, 2020-10-16:
Resolved based on survey and discussion

Vehicle Information Service Specification (VISS) version 2

https://lists.w3.org/Archives/Public/public-automotive/2020Oct/0011.html

closed 2020-10-16 by tguild