w3c/automotive #335: Issues of status codes

wonsuk73, 2020-08-19:
Thanks for good progress of Gen2 Core spec. When I read this spec, I would stand for issues below.

Issue 1.  Does it need to have two separated status codes for invalid token as below? If it does, then reason why?
- 401 (Unauthorized) | token_invalid
- 406 (Not Acceptable) | invalid_token

Issue 2. It seems token_invalid might cover a meaning of toke_missing. If toke_missing is required, then could you explain why?
- 401 (Unauthorized) | token_invalid
- 401 (Unauthorized) | token_missing
- 404 (Not Found) | missing_token
- 406 (Not Acceptable) | invalid_token

Thanks!

UlfBj, 2020-08-20:
Issue 1: 
This is an error, there should not be two status codes for invalid token. The same for missing token (if it should be there at all).

Issue 2:
The invalid token error could be used also for the case of missing token. I guess it is a matter of taste whether there should be a separate error for missing token. I am fine with skipping it if others think so.

wonsuk73, 2020-09-03:
@UlfBj @tguild if there is no objection for #337  and #338, could you merge?

wonsuk73, 2020-09-03:
PRs are merged, so it's done.