w3c/automotive #336: Relationship between VISS and Gen2-Core

wonsuk73, 2020-08-19:
It seems Gen2 Core spec might include Websocket based messaging part as an 5.2. It's totally redundant with VISS spec. Does WG have a consensus for this direction? If it is, I am not sure it is right way.

UlfBj, 2020-08-20:
The Gen2 Websocket protocol is not identical with the VISS Websocket protocol, there are for example several "action" types removed. So it is not redundant, but an evolution of the VISS protocol.
I do not see VISS needs to become a W3C note due to this.

wonsuk73, 2020-08-24:
@UlfBj Thanks for quick relpy.
I checked your point. But I would like to see group consensus more clearly for our direction. I think we might have three options as below.
1) Applying your update into VISS because it’s on the CR stage. And Gen2 - transport only cover HTTPS.
2) Going forward in current context you said above with Gen2 - Core, Gen2 - Transport, VISS
3) Going forward only with Gen2 - Core and Gen2 -transport. And bring VISS into W3C note. Because VISS is covered by Gen2 -transport

I understand as your opinions is 2). But I wonder if Gen2 - transport could cover all of functions of VISS with advanced features, why we need VISS as an standards? What do you think? If am wrong in some point, please correct me.

Thanks!

tguild, 2020-08-24:
@wonsuk73 no need for VISS to be changed to a W3C Note. In fact it should proceed towards final REC stage especially since it is implemented and in use.

wonsuk73, 2020-08-24:
@tguild Thanks! You mean VISS meet all of requirement to quit the CR of W3C. right? If then, why not to move to PR? Do the group have a plan for doing that work within the year?

tguild, 2020-08-25:
Hey @wonsuk73 we can and should advance it indeed. Hasn't been pressing with most attention going into Gen2 and elsewhere but will work with Peter to move it along.

wonsuk73, 2020-08-26:
@tguild Thanks! 👍

closed 2020-09-15 by tguild