w3c/automotive #343: Multiple data points from different parts of tree

UlfBj, 2020-10-22:
As decided on the WG mtg, this issue is opened to discuss the idea of allowing a client to request multiple data points from different parts of tree.

- A client may be interested in multiple data points from different parts of the tree for a specific use case.
- It should be possible to retrieve all data points in one request, instead of having to issue multiple requests. 
* Minimizes bandwidth utilization.
* Ensures that data points are “from the same point in time”.
- Ulf proposed that it could be implemented by augmenting the path expression via a query.
GET Vehicle.Powertrain.BatteryManagement.Battery.AccumulatedChargeEnergy?$pathsEQ[Vehicle.Cabin.Infotainment.Navigation.CurrentLocation.Latitude, Vehicle.Cabin.Infotainment.Navigation.CurrentLocation.Longitude]
- Adnan proposed that it could instead be implemented b using the POST method, and include the extra paths in the payload of the POST request.

After reflecting on the two proposals I now think the solution using POST is to prefer. 
As Adnan pointed out it will make the URL "cleaner". 
And it will also open the door for changing all the queries that are defined in the VISS v2 specification into POST payloads. If we go for introducing the POST method, in both protocols (HTTP and WS), then I think we should also transform the current queries into POST payloads also.

peterMelco, 2020-10-22:
So, is the error handling all or nothing in this case ?

tguild, 2020-11-24:
Ted will see if HTTP2 has a better mechanism for partial response than HTTP1

closed 2020-12-15 by tguild