w3c/automotive #344: Historic data

UlfBj, 2020-10-22:
* If a node supports storing data, then it should be possible for a client to retrieve it.
* Request for retrieval of historic data points for a just passed period of time
** Query with reserved name $history appended to path
** Period syntax as specified in ISO8601
** Vehicle.Powertrain.BatteryManagement.Battery.AccumulatedChargeEnergy?$historyEQPT4M
* Server responds with all data saved for this period.
** Minimum response is the latest value.
* Primary use case: Vehicle temporarily out of coverage.
** No need for client activation/deactivation

peterMelco, 2020-10-22:
The use of persistent storage on vehicles is something that we need to keep optional. Oems needs to be able to estimate life expectancy of a vehicle and have that matched to the number of available writes to storage media.

tguild, 2020-12-15:
I am in favor of historic data but wonder if there should be a mechanism to instruct the service of data that would be desirable, stored subscriptions and data collection occurs regardless of the application running

tguild, 2020-12-15:
Ted's idea compliments and doesn't block Ulf's

UlfBj, 2021-01-20:
Solved in PR#362.

closed 2021-01-20 by UlfBj