w3c/automotive #346: Sampling characteristics control

UlfBj, 2020-10-22:
* The sampling characteristics affect data quality, bandwidth load, etc. 
* Scope of control
** Per signal?
** Per client?
** Per vehicle?
* Per signal
** SET Vehicle.x.y.z#samplemethodX
* Per client
** To be set by - WS:Subprotocol, HTTP: Header
* Per vehicle
** Access controlled node in tree: Vehicle.Connectivity.Data.SamplingCharacteristics
** Node type: Actuator or Attribute, depending on if writeable or not

UlfBj, 2020-10-22:
When this was presented to the WG, one comment was that the Per vehicle option should maybe be left out. 
I would like the OEMs to clarify how the "signal digitalization" system in vehicles is designed. If there is a common sampling frequency, or set of alternative frequencies, then this information would be of interest to a client. It could be available for informational purposes, as an attribute, in the tree. If the Per signal, or Per client cases were to be supported, the information in the tee could then provide the client with the options available.

Another comment was that for the Per client case, the HTTP alternative was of less interest.

peterMelco, 2020-11-03:
So CAN buses have maximum transmission rate per frame/message. They are usually set to be at one rate for system safety. However, one could consider having a sub/pub that have varying frequency that are interfacing can networks with whatever rate but under that constraint. Having frequency per signal that is.

UlfBj, 2020-11-03:
>> So CAN buses have maximum transmission rate per frame/message. They are usually set to be at one rate for system safety.

This is a system characteristics that I think would be helpful to know by a client. Requesting data, either via Get or Subscribe, at a frequency higher than this would not be meaningful, but fractions of it would. 

The model I envision is that the system writes signals from the CAN bus into a "statestorage" at this pace, then the VISS v2 server reads from the statestorage at the pace set by client requests. This decouples the system sample characteristics from the sample characteristics desired by the client. 
Is this model applicable to OEM system implementations? 
If not, how should we model this in a VISS server context?

peterMelco, 2020-11-03:
I think that the answer is complex and is also depending on OEM specifics. If we have nodes that interface with can network(s) which in turn interfaces a ip based network where some other ECU holds the VISS Server, the constraint might not be on the actual can bus but what is allowed from the ECU perspective.

petervolvowinz, 2020-11-10:
If we agree in this information being useful to the client(s). Should we add say sampling freq per signal as an attribute, or would it be part of a general attribute for a branch even for the root of the tree ? 

Per signal is easy to understand from a clients' point view.

If we have a general statement of max freq for all signals below a certain level in the signal tree - The usefulness of this would be use case dependant.

petervolvowinz, 2020-11-24:
Moving to VSS

UlfBj, 2021-01-20:
The dynamic-metadata request described in the CORE specification provides support for the idea presented in this issue.

closed 2021-01-20 by UlfBj