w3c/automotive #357: Scope clarity

samuelweiler, 2020-12-01:
I'm having trouble understanding the scope of the WG.  Example of confusion below.  Would you entirely rewrite the scope section with an eye toward plain language and the reader who isn't in the WG?

Quoting from the charter, emphasis added:

> "This group will develop **service specifications** for exposing vehicle **data** and other information around **vehicle centric functions**."
> 
> "**Services** may include but are not limited to
> 
> - **Vehicle Data**
> [examples]
> -  **named remote procedure functions** defined in a vehicle service catalog (digital key, remote delivery, electric vehicle charging, geofencing)

What, in this context, is a "service" - as in "what service is the WG specifying" when the charter then goes on to define "service" as "data".

And where does this "vehicle service catalog" come from?  and, since the WG is specifying "services", it is specifying this?

Rather than fix this, I suggest a complete rewrite.

See also the mission statement issue: https://github.com/w3c/automotive/issues/352

tguild, 2020-12-02:
The term 'service' is unfortunately overloaded as are many terms at W3C like I noted during Tuesday's Strategy call with CSS Spatial Navigation as we have a Spatial Data on the Web group that is working on... Navigation

The in-vehicle 'service' is the Vehicle Information 'Service' Specification listed in the deliverables, a published spec. The service exposes data/information from the vehicle with examples given. This wording and the mission is largely carried forward from previous charter.

"vehicle service catalog" comes from GENIVI and clarified that with a link.

We will look to refining the wording around service not rewriting and starting horizontal review all over again.

samuelweiler, 2020-12-02:
> This wording and the mission is largely carried forward from previous charter.

I see that.  That doesn't mean it's clear.  :-)

I'm not objecting to "service" being overloaded across the W3C - I'm telling you that I find this usage confusing, in context.

tguild, 2020-12-02:
It is also unfortunately overloaded with GENIVI's choice of using the term in Vehicle Service Catalog, adding to confusion

closed 2021-03-11 by tguild