w3c/automotive #359: How is WG authorizing RPCs?

samuelweiler, 2020-12-01:
What is the WG's approach to authorization of RPCs?  (answering this as a reviewer is tricky since the RPC doc doesn't exist, per this draft charter)  Does the WG need to work on an architecture for that before adopting the specs for those RPCs?

tguild, 2020-12-02:
Tricky to answer as it is not defined yet. We have been evaluating existing RPC models such as gRPC and WAMP plus considering using WoT, all of which support multiple authentication schemes and authorization models. We will contrast the solutions before selecting a path and welcome WebAppSec input in our selection process. Focus is on developing use cases and representing them as named functions in the 'vehicle service catalog' that RPC will use.

samuelweiler, 2020-12-02:
> Focus is on developing use cases and representing them as named functions in the 'vehicle service catalog' that RPC will use.

The RPC work item does not mention "use cases" - it says "the data taxonomy defines service functions available on a vehicle."  That seems to assume that something exists for calling those functions.  (e.g. what's the call flow for setting up the authorization?)

> We have been evaluating existing RPC models such as gRPC and WAMP plus considering using WoT, all of which support multiple authentication schemes and authorization models.

Where is "pick that RPC model" is the work items of the WG?  While working on use cases now is great, security needs to be designed in from the beginning, and it is hard to evaluate the security of the whole without knowing the RPC model.  Generally speaking, picking one model would be good - options lead to holes.  

I reiterate my suggestion for an architectural doc in the work items.

I expect the people in the Web Authn WG and the IETF's OAUTH WG would be more help on this than the WebAppSec WG - WebAppSec's mission is entirely elsewhere.  That said, this WG needs to own its own security architecture.  It should reuse primitives from elsewhere, but the overall security story can't be outsourced.  That might require recruiting participants with appropriate expertise.

tguild, 2020-12-02:
"This specification will follow" is more appropriate for "No Draft" state than "This specification follows" since it does not exist. 

Unclear about how much to state in charter work items such as need to specify work on use cases but as this one is clearly not satisfying, adding the following:

The group has performed some preliminary evaluation of various RPC protocols suitable for connected vehicles, created initial requirements and developing use cases to define explicit functions for the service catalog. Architectural proposals will be contrasted with these examples in order to inform selection of RPC protocol, including authentication and authorization means.

Although too early for RPC authentication/authorization but we will consider the recently added (end of Summer) auth in second version of VISS. <tangent/>That document hasn't had its FPWD yet and will want security review. We are fortunate to have attracted some expertise in that front and welcome wider review.

tguild, 2020-12-02:
https://w3c.github.io/automotive/spec/Gen2_Core.html#access-control-model

tguild, 2020-12-15:
Came up with agreed upon wording with Sam for charter for this issue

"The Working Group will document the overall RPC architecture, including its authorization mechanism."

closed 2020-12-15 by tguild