w3c/automotive #360: Error handling of requests addressing multiple nodes.

UlfBj, 2020-12-02:
As discussed on the WG meeting, the Filtering chapter has a chapter describing the error handling when one or more of a set of addressed nodes are found in the VSS tree. 
The chapter does not describe the details, but says that an error code is returned together with information about which node(s) that are invalid.

UlfBj, 2020-12-02:
My view on this is that the response shall not contain any data for the eventual nodes that are valid, but only error information related to the invalid nodes/paths. 
The argument for this is that for subscription requests such a behavior seems to be the right, as otherwise a subscription session is started that will return an incomplete set of data points, compared to the subscription request. If this is the most appropriate server behavior for subscription requests, then the same pattern should be used for get requests.
I think that the error code 404 Not Found is appropriate, together with a payload containing the invalid paths. 
With the current spec error response format, for WS it would be something like
{"error":{"number":"404", "reason":"Not Found", "message":"Invalid paths: path1,..,pathN"}
This might for the message part maybe not be so machine readable, if that is a problem we might need to change our error response format.

tguild, 2020-12-02:
After talking with @ylafon realize my thought of giving partial results and error code is flawed. It should either fail entirely providing an error code and body/resource payload providing details of what failed why or embed error per node/tree branch that failed or was blocked (authorization).

Yves expressed it is probably cleanest to separate the requests instead of combining them and how much more efficient HTTP2 is. I did not explain we are trying to handle WebSocket and HTTP requests consistently.

tguild, 2020-12-15:
Resolved: if any path is invalid, fobidden etc the HTTP response code should respond accordingly and not return partial results

UlfBj, 2021-01-20:
Solved in PR#362.

closed 2021-01-20 by UlfBj