w3c/automotive #361: Response syntax for multiple data points

UlfBj, 2020-12-03:
The filtering functionality makes it possible for a client to formulate one request which will lead to a response containing multiple data points, and the spec should define a syntax for such responses. There are four variants that may occur:
1. One signal, one value: E.g. a get request addresses one node in the tree.
2. One signal, n values: E.g. a subscription request addresses one node in the tree, with curve logic processing.
3. m signals, one value per signal: E.g. a get request addresses multiple nodes in the tree.
4. m signals, n values per signal: E.g. a subscription request addressing multiple nodes in the tree, with curve logic processing. 

My proposal on a syntax for theses four alternatives are shown below.
1. One signal, one value:
```
“datapackage”:{“path”:”X”, “datapoint”:{“value”:”Y”, “ts”:”Z”}}
```
2. One signal, n values:
```
“datapackage”:{“path”:”X”, “datapoint”: [{“value”:”Y1”, “ts”:”Z1”}, …, {“value”:”Yn”, “ts”:”Zn”}]}
```
3. m signals, one value per signal:
```
“datapackage”: [{“path”:”X1”, “datapoint”:{“value”:”Y1”, “ts”:”Z1”}}, …, {“path”:”Xm”, “datapoint”:{“value”:”Ym”, “ts”:”Zm”}}]
```
4. m signals, n values per signal:
```
“datapackage”: [{“path”:”X1”, “datapoint”: [{“value”:”Y11”, “ts”:”Z11”}, …, {“value”:”Y1n”, “ts”:”Z1n”}]}, …, {“path”:”Xm”, “datapoint”: [{“value”:”Ym1”, “ts”:”Zm1”}, …, {“value”:”Ymn”, “ts”:”Zmn”}]}]
```
I see this as being described in a separate sub-chapter in the Filtering chapter.

tguild, 2020-12-08:
started discussing in WG call https://www.w3.org/2020/12/08-auto-minutes.html#item02

Gunnar encourages alignment with value exchange/measurement formats

https://at.projects.genivi.org/wiki/display/DIRO/Value+measurement+formats

UlfBj, 2021-01-20:
Solved in PR#362.

closed 2021-01-20 by UlfBj