w3c/automotive #364: VISS version 2 review

tguild, 2021-01-20:
Please use this issue for minor suggested changes or fixes (eg typographic errors) to the VISS version 2 specifications. Substantive changes instead warrant creation of new issues so they can discussed further.

https://rawcdn.githack.com/w3c/automotive/64f3a1cfdd53ccb8411ccabb52e1cc39a7148d32/spec/VISSv2_Core.html

https://rawcdn.githack.com/w3c/automotive/64f3a1cfdd53ccb8411ccabb52e1cc39a7148d32/spec/VISSv2_Transport.html

peterMelco, 2021-02-02:
**VISS v2 CORE review**

No major issue in core what I could see.

General comments that applies to CORE:

Why are we defining our own set of rules when it comes to capitalisation of certain words such as Client , Purpose , Short term.  I believe that this makes the text harder to understand since takes away the focus of the actual context. My suggestion is to use capitalisation for acronyms and stick to normal written English where we use capitals for proper nouns and in the beginning of sentences.  

In addition to this the use of cursive and capital letters for MAY,SHALL , MUST looks ok.

**Abstract**:

In the abstract we should write what VISS v2 actual stands for. 

Remove pharenthesis “… ,e.g  navigation, media and vehicle data.”

**5.1.3**

“Purpose: Get asynchronous messages containing the value(s) addressed by the path. The triggering rules for issuing the notification messages are set by the filter data.
“

The use of notification messages without defining it as the actual asynchronous messages leaves the reader with a sense of “ what is that ” … it is also so that the notion of notifications is commonly used for pushing messages to a driver which has little to do with the actual vehicle sensor data so an explanation early on would be helpful.

The concept triggering rules also need a definition.
I would rephrase so that async messages and notifications is understood to be the same thing.

Suggestion:
Asynchronous messaging containing values adressed by the path. The messages can be viewed as notifications, which are triggered by rules defined by the filter data.

**5.1.4**

We are using associated to which seems to not as common as using associated with throughout the document. This is ok but a bit weird language I think…

**5.1.5**

Just write “ Asynchronous client notification” remove according to bla bla…

Should the notification  include “ HEY a triggering rule was met “ or what is this notification…

Timestamp . What do we mean by trigger activation ? When the signal is sent from the server or when we issued a trigger which could be viewed as the activation.

Defintion list:

Add notification and triggering rule.

**6 Filter request**

Add definition reference for curve algorithm: https://www.geotab.com/blog/gps-logging-curve-algorithm/

**7. Metadata Request**

Don’t start with writing what it is not doing. Rephrase:

Metadata requests are used when the client wants to retrieve meta data associated with the VSS node. Metadata such as a signal description , unit and type.

Let the Vehicle Signal Specification point to vspec files , perhaps.

**8.4.4** 
remove etc from “ …Device and App roles may involve MAC address checks, message round-trip measurements, etc. “ or write what is included in etc.

**8.8** 

VSS layering concept needs reference link or definition. Could not find one my self though.

**List of figures**

Add scope list and purpose list

Reference  is list empty

UlfBj, 2021-02-08:
> Why are we defining our own set of rules when it comes to capitalisation of certain words such as Client , Purpose , Short term. I believe that this makes the text harder to understand since takes [...]

The Client with capital C is defined in ch 8.4.1. When in the Access control chapter Client is written it refers to such a client, and not a generic client. So I think for at least that example it is an acceptable usage. The first occurrence of Client should link to ch 8.4.1. If it is the case for others needs to be checked.

peterMelco, 2021-02-08:
Well, I don’t really agree. I still not see an argument for us defining new language concepts.

PeterWinzell, 2021-02-09:
Cool with current suggestion from TED.

UlfBj, 2021-02-23:
Latest versions of documents per 210223:
https://rawcdn.githack.com/w3c/automotive/3b84ac8b41586c1dbfc83aed5ca956a77a9f1445/spec/VISSv2_Core.html
https://rawcdn.githack.com/w3c/automotive/3b84ac8b41586c1dbfc83aed5ca956a77a9f1445/spec/VISSv2_Transport.html

tguild, 2021-06-15:
Issuing call for consensus at today's working group meeting, you can indicate your support with a thumbs up emoji or comment

closed 2021-12-07 by tguild