w3c/automotive #365: Fragments are not transported over HTTP

UlfBj, 2021-01-22:
The chapter "Metadata Request" in the CORE spec says that the syntax uses a fragment component as shown below
HTTP GET
URL: Vehicle/ADAS/ABS#static-metadata (or dynamic-metadata). 
The problem with this is that HTTP removes the fragment before issuing the request. 
Information about it can be found here
https://blog.httpwatch.com/2011/03/01/6-things-you-should-know-about-fragment-urls/
I would therefore like to propose that the request instead uses the query syntax as shown below.
HTTP GET
URL: Vehicle/ADAS/ABS?metadata=static (or dynamic).

peterMelco, 2021-02-02:
Look good to me...actually I like this more than the previous suggestion.

tguild, 2021-02-16:
Using querystring instead

closed 2021-02-16 by tguild