w3c/automotive #369: MQTT as transport protocol

UlfBj, 2021-02-23:
A proposal on how to add MQTT as a 3rd VISSv2 transport protocol is described here
https://github.com/MEAE-GOT/W3C_VehicleSignalInterfaceImpl/tree/master/server/mqtt_mgr

To be precise, a thin application level protocol is added on top of MQTT to enable it to transport any VISSv2 payload. Through this addition on top of MQTT a complete abstraction of MQTT is possible, so that an application level client do only have to know the VISSv2 payloads to be able to communicate with the VISSv2 server via MQTT. It must also be able to obtain the vehicle identity through some out-of-band method (i. e. it cannot be found initially over MQTT). 

Adding this to the specs would not change the CORE document, more than mentioning MQTT as one of the supported transport protocols. The TRANSPORT document would be augmented with an MQTT chapter, and sub-chapters, following the pattern for the already supported transport protocols, plus a separate sub-chapter explaining the application level protocol.

petervolvowinz, 2021-02-23:
I think that we should use TLS over MQTT. There are levels of authentication which we could specify as well, e.g if the client should provide anything for the server or not.

UlfBj, 2021-11-19:
Fixed in PR427

closed 2021-11-19 by UlfBj