w3c/automotive #380: Feature: Payment endpoint, allowing payment in a connected car via a standart endpoint

dkdndes, 2021-04-23:
Within the car, it's assumable that payments will need to take place. Some vehicles already include it, but it does not exist in a standardized and easily consumable way. W3C develops at the moment for Browser standard payment processing solutions, this will reduce costs and time for the development of the appropriate customer journeys.

The feature requests suggest a similar approach for automotive payments, to allow reduced costs and easier development of payment solutions in the connected car. 

Is this something the working groups would see within its scope?

tguild, 2021-04-26:
@dkdndes we had a task force for awhile to explore W3C Payments API for automotive use cases

https://github.com/w3c/automotive-pay/wiki

short version: there are some small gaps such as getting a merchant payment URI to the vehicle while trying to mitigate fraud (POI and data points we can provide from the vehicle would help) but its doable. We had three major payment providers interested in prototyping if we could get an automotive manufacturer to agree to play. We did not succeed, so shelved it. Most of our contacts are within connected vehicles, telematics data divisions and only got meaningful conversations with one manufacturer's in-vehicle payment processing engineers. At the time they were unfortunately more focused on a one off solution for a single gasoline company. 

We get occasional inquiries and I point out how it is still feasible and we can resume if we get an auto manufacturer interested.

dkdndes, 2021-04-30:
@tguild we run a bunny race eventually selected a startup, and  DAX payment provider in Luxembourg were set up/founded. Mercedes did a similar thing. Simple reason, why have a payment provider as a man in the middle. Beamer did the same, and in all cases, they do use it for the current subscriptions (like special maps, traffic info, ...). 

It's not yet used as part of the shared economy use cases or for the in-car "butler" functionality use cases - if you mean this with not many "use cases". 

And due to this, the Rest API exists also, as you can imagine, and the connection problem (TCP/IP in a Tunnel, e.g. in the swiss alps) is solved, too. 
My issue is that no standard API was defined so far, and adopted by all companies. If that would be the case, it would open a whole universe of solutions (use-cases) by start-ups that could make money on and with the car platform (which in the end the car is, an SW platform from a startup point of view. 

I requested the feature because I work on the start-up side of things meanwhile, and not within the car company context anymore. I see here a gap between the market requirements and the available options.

p.s.

And I might remember Shell, with Jaguar in the Youtube Video at a fuel station (2016); assume that's what you mean?

UlfBj, 2021-11-19:
An active standardisation project can be found here:
https://at.projects.genivi.org/wiki/pages/viewpage.action?pageId=65896819

closed 2021-11-19 by UlfBj