w3c/automotive #382: Access Control levels

tguild, 2021-04-27:
In a breakout call today on VISS v1 vs v2, there was expressed a desire to keep access control design simpler. For in-vehicle applications the environment would be more rigidly monitored and controlled so therefore more trusted. Allowing for devices (eg smartphones or tablets) to be paired and run applications from or applications running in the cloud permitted to connect to the vehicle should warrant more stringent security.

We may reach a compromise and meet in the middle or end up with two different approaches yet should be clear on what they can have in common so ideally a client application can handle either without too much special handling. 

Please elaborate on sspects that are considered complex, propose alternatives and let's identify commonalities.

SebastianSchildt, 2021-05-04:
Similar to #381  I would argue for some _options_ here

I am not perfectly familiar with all details of what is currently possible with the VISS V2 access control definition, but here is my view wrt  access control and authorization in VISS:

Depending on the deployment, when using VISS in a vehicle, there might even be situations where you run in a "trusted" environment, and my not need to authorize at all. (or simply  using TLS for client authorization is enough). So maybe a *base profile* VISS can even exclude any access control on VISS protocol level

Most of the time, in most deployments, you would probably want some form of authorization & access control.  Here I feel it is important that it does not demand any specific security architecture or model.

To not become as generic as VISS V1 ("add security here"), the spec should still have some generic technical requirements such as using JWT tokens in authorization.

The VISS specification should define a "flat" way of defining permissions without relying on some "roles" or "policy databases" defined elsewhere. Just specify how a basic JWT token should look like containing claims for read or write access to any VSS Path. (Full disclosure: That is the model kuksa.val is following)

Even though we are not using it on VISS level currently,  I concede it might also useful to define how tokens should look like with Role-based access models, which might be used in some deployments. 

Personally, I do consider stuff such as which server grants what token and how long they live as very out of scope for a protocol specification. . Not sure how W3C standards work, but in IETF such ideas would probably go into a separate "informational" RFC.

tguild, 2021-06-01:
@erikbosch and @SebastianSchildt on last week's call @UlfBj wondered if we could design the auth piece to scale down between the more complex and simpler, less access grant server, transparently to the client.Isaac Agudo (not finding his github handle) was on the call and agreed, thinking we could meet in the middle. I thought an implementation might want to support both, simpler for more trusted (eg OEM) applications and complex for third party apps which might need permission per driver, trip or other parameters that could be influenced externally or if permitting a device such as a smartphone to pair to VISS running on the vehicle.

https://www.w3.org/2021/05/25-auto-minutes.html

Curious your thoughts, also consider attending today's call if you're available and rather talk than type

UlfBj, 2021-09-28:
>> a "flat" way of defining permissions without relying on some "roles" or "policy databases"

An alternative design that meets these requests are proposed here. 
The spec will besides the current access control model also describe an alternative model where the "normative description" (formally being optional) starts with the access token. That is, how the access token is obtained is not specified. 
The access token claims differs from the existing ones in that the claims "scp" and "clx" are removed, and replaced by the claim "sac" (signal access), which has an object as value, as exemplified below.
```
"sac": [
    {"path": "Vehicle.Powertrain.FuelSystem.Level", "accmode": "read-write"}, 
    {"path": "Vehicle.Powertrain.FuelSystem.Range", "accmode": "write-only"}
]
```
A server can be implemented that supports both alternatives. In that case, when it finds that the access token contains a "sac" claim, it will search the Purpose-list policy document for a purpose that contains all signals in the "sac" claim, and access mode restrictions that validate the claim. If so, then the request is granted. This means that this client logically is given a "super role" that is granted for all purposes in the Purpose-list. The server SHALL use different keys for signature validation of the access tokens used in this alternative model, and in the existing model. 
A server that only supports this alternative is free to implement the token validation without involving a Purpose-list.

SebastianSchildt, 2021-11-09:
If I understand the proposals correctly, I would prefer @UlfBj  proposal. While the one  by @isaacagudo might be more similar to what is already there, it also shares the same challenges: Its very complex, and assumes a certain "system architecture", which probably only applies to a (small?) subset of use cases, so users/implementers will still do "non-standard" workarounds.

By just allowing to use the sac directly, you still have a fully specified _protocol_, but the only required parties are "client" and "server", which I personally prefer to a protocol that _requires_ more than two involved parties. 

I understand from @isaacagudo that there might be a case where a server/system (for whatever reasons) might not want to permit access to everything specified in the sac. This might happen, but I would consider this application/deployment specific, and it might be acceptable for a server to return 401 when a path is read, even if it is in a sac. For a client side, you need to deal with "errors" anyway, a complex mandatory handshake before communication, does not make things easier for implementers.

Just my two 🪙

isaacagudo, 2021-11-09:
It is true that the proposed flow requires a companion server for the VISS server, i.e. the Access Token Server, but only to implement the access control logic outside of the VISS server, following the current trend for access control in HTTP APIs.
Access control policies are managed by the ATS, not the VISS server. This way the VISS server implementation is very simple, in the sense that it only need to check that the Access Token is valid and that the requested signal matches the sac claims.
If you don't want to involve any external party, i.e. only client and server, you either avoid implementing security at all or you need to pair them beforehand, using something similar to Bluetooth pairing/bonding.

tguild, 2021-11-09:
Isaac provided this link in Slack

https://www.pingidentity.com/en/company/blog/posts/2020/everything-need-know-api-security-2020.html

tguild, 2021-11-30:
resolved in https://github.com/w3c/automotive/pull/435

closed 2021-11-30 by tguild