w3c/automotive #385: VISS Base vs. Extensions

erikbosch, 2021-05-05:
In #381 and #382 it is discussed if certain parts of VISS should be optional. Would it be an idea to have a generic concept/syntax where certain parts of VISS can be specified as mandatory (base functionality) while others are specified as optional extensions. Somewhat similar to https://en.wikipedia.org/wiki/RISC-V#ISA_base_and_extensions

That could possibly be useful for the future when organizations start offering VISS implementations to possibly customers (both clients/applications and server-solutions) to have a common language to describe what is supported. E.g. a syntax to specify that a VISS server implementation for example supports Authorization and PathFilter but not CaptureFilter

UlfBj, 2021-05-25:
I agree with what has been said above. So the next step would then be to 
1. exactly specify what should be optional (all not optional is mandatory).
2. Decide how a client queries the server on its capabilities.

I would propose that 2. is solved by using the "static metadata" request that the spec supports (CORE ch 7.4), by adding
"server-capability" to the list of supported key names. 
An HTTP request would then look like below.
GET /Vehicle?filter={"op-type":"metadata", "op-value":"static","op-extra":"server-capability"}   HTTP/1.1
The outstanding question would then be how the server expresses its capabilities?
Maybe the different optional parts in the spec can be tagged (shown in the spec), and the server then replies with a list of the tags that are supported?

UlfBj, 2021-12-15:
Fixed in PR#437

closed 2021-12-15 by UlfBj