w3c/automotive #389: Issue on error response example in VISSv2 Transport spec

wonsuk73, 2021-07-15:
When I review the [VISS version 2-Transport](https://raw.githack.com/w3c/automotive/gh-pages/spec/VISSv2_Transport.html) spec, I found that error response examples should be revised to sync the section 4.1 Status Codes as well as  the section 5.3.2 JSON Schema Definitions for whole doc.

For example, there is an error response example for section 5.1.2.1 Read as below.
```
              HTTP/1.1 404 Not Found
              Content-Type: application/json; charset=utf-8
              Date: Tue, 13 Jun 2014 19:47:27 GMT
              Connection: keep-alive
              Transfer-Encoding: chunked
              {
                 "message": "No match for path in tree."
              }
```

It should be changed with as below according to the section 4.1 Status Codes and the section 5.3.2 JSON Schema Definitions
```
              HTTP/1.1 404 Not Found
              Content-Type: application/json; charset=utf-8
              Date: Tue, 13 Jun 2014 19:47:27 GMT
              Connection: keep-alive
              Transfer-Encoding: chunked
              {
                 "error": {"number": 404, "reason": "invalid_path", "message": "The specified data path does not exist."}
              }
```

Am I right? I already made the PR for this change as an example. If there is the consensus for this within the group, I am going to review all of error response examples fo the doc and make PRs to apply this changes for the whole doc.

wonsuk73, 2021-07-15:
@tguild, @UlfBj, @peterMelco, Could you review this?

UlfBj, 2021-07-16:
@wonsuk73 You are right, thank you for catching it. 
I have merged the PR you created, and would be very grateful for help to fix other similar errors.

wonsuk73, 2021-07-16:
@UlfBj Thanks a lot for quick review! I am going to fix the others for whole doc! 👍

wonsuk73, 2021-07-16:
@UlfBj Could you review PR #390?

wonsuk73, 2021-07-21:
Thanks @UlfBj
I think this issue had been solved, so let's close it.

closed 2021-07-21 by wonsuk73