w3c/automotive #393: Simply HTTPS message examples on VISSv2 Transport spec

wonsuk73, 2021-07-27:
VISSv2 Transport spec include many message examples and these message has big HTTP header information which is not very meaningful for the spec. Below is an one of example. 

Example: Request:

              GET /Vehicle/Drivetrain/InternalCombustionEngine/RPM   HTTP/1.1
              Host: 127.0.0.1:1337
              Connection: keep-alive
              Accept: application/json
              User-Agent: Chrome/34.0.1847.137 Safari/537.36
              Accept-Encoding: gzip,deflate
              Accept-Language: en-US,en;q=0.8,de;q=0.6
              
I would like to suggest to change this like below to make more concise and clear. It is to keep only meaningful information in the example on the spec.

              GET /Vehicle/Drivetrain/InternalCombustionEngine/RPM   HTTP/1.1
              Host: 127.0.0.1:1337
              Accept: application/json
              ...

I made the PR(#392) as a sample to apply this to the VISSv2 Transport spec. If the gorup agreed on this, I will apply this idea into the whole spec.

wonsuk73, 2021-07-27:
@UlfBj @peterMelco @tguild Could you share your view on this?

peterMelco, 2021-07-27:
I think your comment makes sense.

SebastianSchildt, 2021-07-27:
I like it! It removes some noise from the spec.

UlfBj, 2021-07-27:
I also like the idea. With three positive out of three I say we go for this. I will merge #392, and we will be very grateful if @wonsuk73 will update other occasions of this.

wonsuk73, 2021-07-27:
@UlfBj @peterMelco @SebastianSchildt Thanks! I am going to apply this idea into the whole doc.

wonsuk73, 2021-07-27:
@UlfBj Could you review the PR(#394)? Which is about applying this idea to the rest of the doc.

closed 2021-09-15 by UlfBj