w3c/automotive #409: Filter request syntax simplification

UlfBj, 2021-09-10:
The filter expression currently consists of three key-value pairs as shown below. After feedback from a client implementer that this leads to long and hard to read requests, an analysis lead to that the expression can be simplified to two key-value pairs instead.
Current:
{"op-type":"<x>", "op-value":"<y>", "op-extra":{<z>}}
New proposal:
{"type":"<x>", "value":"<y>"}

Below follows an example of the current syntax, and the proposed syntax.
Current:
{"action":"subscribe","path":"Vehicle/Cabin/Door/Row1/Right/IsOpen","filter":{"op-type":"capture","op-value":"time-based","op-extra":{"period":"3"}},"requestId":"246"}
Proposed:
{"action":"subscribe","path":"Vehicle/Cabin/Door/Row1/Right/IsOpen","filter":{"type":"time-based","value":{"period":"3"}},"requestId":"246"}

crea7or, 2021-09-10:
a good idea, just realized that metadata requests too complex and can be changed to:
```
"filter": {
 "type": "static-metadata", 
 "value": ["prop1", "prop2"]
}

"filter": {
 "type": "dynamic-metadata", 
 "value": ["prop1", "prop2"]
}
```

wonsuk73, 2021-09-14:
Basically I agreed with the proposal from @UlfBj. 
In specific there are two filter functions which use a "op-extra" property like capture and metadata filter. 
#1. Regarding metadata filter function, I support @crea7or's proposal.
#2. Regarding capture filter function, I would like to suggest names of the type as below to make clear the type's meaning. 
- time-based-capture
- range-capture
- change-capture
- curve-logging-capture

UlfBj, 2021-09-27:
Fixed in PR418

closed 2021-09-27 by UlfBj